AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #366: Strange behavior of Discharge and Calibration functions in clamshell mode

teplynin, 2021-08-28:
I am using AlDente Pro 1.07 on a Macbook Air M1 with an LG Ultrafine 4K external monitor and Magic Keyboard. OS Big Sur 11.5.2.

If I lock the computer and wait for the external monitor screen to turn off (~ 5 sec), the monitor stops turning on when I press the Magic Keyboard keys until the laptop lid is opened. The keyboard itself works, Caps Lock lights up.

If I close the program or turn off the Discharge mode, everything works correctly again: when the computer is locked, the monitor fell asleep, pressed any key - the monitor woke up. 

Reinstalling AlDente,  restarting the computer and reset power settings did not help.

snarf55, 2021-09-06:
same problem here. MBA M1, Big Sur 11.5.2, Aldente pro 1.07

teplynin, 2021-09-11:
UP.
Look like support is dead

nuggie, 2021-10-02:
same issue here ... BenQ Monitor attached via Thunderbolt Dock over HDMI

teplynin, 2021-11-08:
Well, problem still exists in MacOS Monterey with AlDente 1.09.  Wasted money for pro version.

MatthiasKerbl, 2021-11-15:
Hi everyone,

We have just released a new version of AlDente (1.10). https://github.com/davidwernhart/AlDente/releases
This version should fix the issues you have experienced with AlDente.

If you still have problems with AlDente, please feel free to let us know.

Thank you for your patience!

Cheers,
Matthias

closed 2021-11-15 by MatthiasKerbl