AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #367: MacBook is not charging after AlDente was uninstalled

VladVanHellsing, 2021-09-01:
Hey guys. I facing an issue after AlDente was uninstalled. It somehow "jammed" the charging process and MacOS thinking that there is no need to charge the machine. Please advice! Thank you!

**Example:** 
1. I charged the machine up to 100%
2. I uninstalled AlDente
3. I used MacBook without charge to low the battery down to 74%
4. I plugged in the charging cable

**My specs:**
Hardware: MBP 16 (2019)
OS: BigSur 11.5.2

**How I uninstall AlDente:**
- Open the app
- move the slider up to 100%
- uninstall helper from settings
- uninstall the app
- restart machine

Please see screenshots in the attachments.

Possible solutions:
- Install Aldente again
- Play with charging slider, untill Mac begins to charge again.

[image]

[image]

[image]

burnout-syndrome, 2021-09-01:
Did you perform a SMC reset after uninstall?

VladVanHellsing, 2021-09-01:
> Did you perform a SMC reset after uninstall?

Hi! No, I did't. Will try.

closed 2021-09-01 by VladVanHellsing

boafur, 2021-09-06:
Hello! I have experienced this issue, and after restarting the MacBook and performing an SMC reset multiple times, the problem does not seem to go away. The only way I am able to charge my MacBook is by installing AlDente again and messing with the slider until the MacBook charges.

I am using a MacBook Pro 2016 on macOS Monterey Beta 6 with AlDente version 1.07.

VladVanHellsing, 2021-09-06:
I think the issue is still there for MBP 16.

I did another spin with AlDente (after resetting SMC, so AlDente should restore the "clean" SMC setting after been uninstalled) and faced the same results (and reset SMC again to let my Mac charge again).

boafur, 2021-09-06:
It seems that you are able to allow your mac to charge again without AlDente by resetting the SMC, but I have tried resetting the SMC, PRAM, NVRAM, and many other combinations but to no avail.

VladVanHellsing, 2021-09-07:
In this case, I can recommend reinstall macOS (to clear SMC and the rest of the system services) and wait until AlDente received full Monterey support after its release.

I can't say why the same machines with just different OS have a completely different reaction to a "hardware" SMC reset.

unphased, 2022-07-18:
I'm experiencing an issue with my Apple Silicon 2021 Macbook Pro and although I use Al Dente, I believe it is unrelated to Al Dente. It is described here, and many others have the issue: https://www.reddit.com/r/mac/comments/uy84ge/16_m1_max_mbp_wont_charge_until_reboot/

I am installing the 12.5 beta now since the fix is confirmed to be in it. Just leaving this note here so others can follow the trail.