AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #368: What is supposed to happen?

gbreeze64, 2021-09-01:
Open the software on my 13" M1 MacBook Pro and nothing? No dialogs, no menubar icons, not a thing?  Weird.

darelldd, 2021-09-05:
Yeah, same here. I got a warning that it'll be running in Intel mode so I won't have pro features. And then.... well, I seem to have NO features. I can't see a way to actually use the app. When I try to restart the app, it tells me nope! The app is already running. So, I give up. Where might it be hiding?  Late 2013 MBP

MatthiasKerbl, 2021-09-19:
Hi @gbreeze64 and @darelldd,

When you open AlDente Free there should be this icon:

[image]

in your menu. Click on it to change the charge limit.

Cheers,
Matthias

darelldd, 2021-09-19:
Yes! There it is up at the top! 

This would be super info to include in the notes somewhere. For the life of me I couldn’t see WHERE it was running as a physical icon.

-= Darell =-

closed 2021-09-21 by MatthiasKerbl