AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #369: Discharge function not working / charge limiting not working

brypwn, 2021-09-02:
Hi, I'm currently using an M1 MacBook Air. With Al Dente on, I set the charge limiter to 90%. My battery was at 100%, so I used the discharge function to run the battery down to my desired charge level. However, the Macbook stops charging for a few seconds, and immediately charges again. This goes on and on until I disable the discharge function, or unplug the charger.

After I removed the charger and let my Mac go down to about 85%, I tried the charge limiter function, set at 90%. After plugging in the charger, it charges to 90%, stays there for a few minutes, then goes above the limit, charging to 100% in the end.

Any idea what's going on?

Thanks!

brisk84, 2021-09-03:
Version 1.07 works unstable on my Air 2020 M1 too. Trying uninstall, reinstall, reboot...
Then back to AlDente Classic 2.2 and it works fine!

vividfog, 2021-09-04:
I've got an M1 Air as well, and the story is slightly different. Trying to capture the steps here. It's unclear to me, how Discharge in Clamshell Mode is _supposed_ to work with M1 Air. Maybe the developers could elaborate? 

1. Clicking on Discharge makes the external screen go black, as if the computer is going to sleep, or the display was turned off. Touching any key or trackpad wakes the display up and I can continue working. But no discharge.

2. After this, clicking Discharge again shows a "Power Plus" icon, which means it's charging, verified with Coconut Battery (charging, 18 Watts). Clicking again and again makes it cycle between "Power Pause" and "Power Plus" icons, but at no point I see "Power Minus", which I'm expecting.

3. Charging stops at 80%, the level I set it up to use. Clicking Discharge after this shows a blue bar, toggling the Sailing icon on/off. At this point Coconut Battery says "charging with 0 Watts". I don't know how to verify whether it's actually "sailing", but the percentage seems stable.

So yeah, it behaves in strange ways and in multiple steps. But I can't get it to _actually discharge_. Any advice? Any debug steps I could try?

Setup: Macbook Air M1 + LG 5k via Thunderbolt 3, plus lots of other peripherals, AlDente Pro, latest version

MatthiasKerbl, 2021-11-05:
> Hi, I'm currently using an M1 MacBook Air. With Al Dente on, I set the charge limiter to 90%. My battery was at 100%, so I used the discharge function to run the battery down to my desired charge [...]

Hi @brypwn,

Do you have Apple's "Optimized battery charging" disabled?

Cheers,
Matthias

closed 2021-12-11 by MatthiasKerbl