AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #376: MacBook still charging while powered off

martin-braun, 2021-09-11:
Hi, Pro user here. Thanks for this great software. 

I told Al Dente to avoid charging while sleeping or being powered off, but it seems to keep charging when powered off.

[image]

[image]

Why it keeps charging when powered off, please? Sure I can disconnect when I power off, but this would be nice if it was working. :)

IamCrypt0n, 2021-09-17:
I can confirm this behavior. Bought this PRO version just for this feature but it is not working on my M1 MacBook :(
[image]

[image]

joujek, 2021-09-21:
+1 for this feature - I'm PRO user

MatthiasKerbl, 2021-09-21:
> Hi, Pro user here. Thanks for this great software.
>
> I told Al Dente to avoid charging while sleeping or being powered off, but it seems to keep charging when powered off.
>
> [image]
>
> [image]
>
> Why it [...]

Hi @martin-braun,

could you describe in detail what happens when you turn off your MacBook?

Cheers,
Matthias

MatthiasKerbl, 2021-09-21:
> I can confirm this behavior. Bought this PRO version just for this feature but it is not working on my M1 MacBook :(
> 
> [image]
> 
> [image]

Hi @IamCrypt0n,

due to hardware changes of the M1 MacBooks, the feature "Stop charging while powered off" is not possible on Apple Silicon MacBooks. Hence, the feature is called "Stop charging when app is closed".

Cheers,
Matthias

martin-braun, 2021-09-24:
Seems to work now, not sure why is that. If I experience it again, I will reopen a new issue after inspecting further.

closed 2021-09-24 by martin-braun