AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #381: Stops charging to limit after sleep

mehmet-safak, 2021-09-13:
M1 MBA here. After a few times sleep&wake up, the Mac doesn't charge to the set limit. This is super buggy right now

MatthiasKerbl, 2021-09-21:
> M1 MBA here. After a few times sleep&wake up, the Mac doesn't charge to the set limit. This is super buggy right now

Hi @mehmet-safak,

Do you use AlDente Free or AlDente Pro?

Cheers,
Matthias

mehmet-safak, 2021-09-23:
Hi @MatthiasKerbl 

I'm using AlDente Free. In this condition, I can't see me upgrading.

MatthiasKerbl, 2021-09-25:
> Hi @MatthiasKerbl
> 
> I'm using AlDente Free. In this condition, I can't see me upgrading.

Which macOS version do you use?
Could you explain the issue you encounter with AlDente Free in more detail?

theArtechnology, 2021-10-20:
I can confirm this, I use Al Dente Pro. I use version 11.6.

MatthiasKerbl, 2021-11-05:
Hi @theArtechnology and @lamemair,

AlDente can not control the charging behavior while your MacBook is in sleep or powered off. However, AlDente Pro has two features to avoid charging to 100%. We have a detailed explanation for those features on our blog:

- https://apphousekitchen.com/feature-explanation-stop-charging-when-sleeping/
- https://apphousekitchen.com/feature-explanation-stop-charging-when-powered-off-app-closed/

Cheers,
Matthias

closed 2021-11-05 by MatthiasKerbl