AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #382: Charging beyond the set limit

Drallas, 2021-09-14:
AlDente always worked perfectly and charged to the set limit and then stopped. 

With the latest version on two new M1 MacBook Air(s) it's charging far beyond the limit I set. (limit 80% stopping at 91%)

I removed and reinstalled the helper App a couple of times but it seems not to respect charge limits anymore!?

danrap2, 2021-09-17:
Hi!
seems happened to me; m1 mac book air, charge limit to 70%; after a couple of days in "stop"state, it charges to 80%

brisk84, 2021-09-17:
I'm too back to v.2.2, it's work perfectly!

Oll0w4in, 2021-09-21:
Same problem here, just bought AlDente Pro two weeks ago and now it is working way worse than AlDente Classic. Very sad

kaandok, 2021-10-05:
I also have this issue.

Using Setapp version 1.07, M1 MBP.
Charging limit set to 80%.
Icon shows charging is paused but battery gets charged well above 80% (although this is very slow and gradual)

I reinstalled the helper a couple times, didn't help.
I ran calibration a couple times, didn't help.

MatthiasKerbl, 2021-12-11:
> AlDente always worked perfectly and charged to the set limit and then stopped. With the latest version on two new M1 MacBook Air(s) it's charging far beyond the limit I set. (limit 80% stopping at 91%) I removed and reinstalled the helper App a couple of times but it seems not to respect charge limits anymore!?
> 
> PS : For now I have reverted back to v2.2. Bummer because I like the Pro version and it's features; but before I get it, I want to make sure all works as expected!)

Hi @Drallas,

Do you still encounter this issue with our newest version AlDente 1.11?
Was your MacBook turned off or in sleep when it charged past the charge limit? Since AlDente Free can not control the charging behavior of your MacBook at all while it is in sleep or turned off. You can read more about this on our blog:
https://apphousekitchen.com/feature-explanation-stop-charging-when-sleeping/
https://apphousekitchen.com/feature-explanation-stop-charging-when-powered-off-app-closed/

Cheers,
Matthias

Drallas, 2021-12-12:
> 

@MatthiasKerbl It seems to be fine for a while, my charge limit is set to 90% and when it's charing it stops there now. I charge with the lid closed.

closed 2021-12-12 by MatthiasKerbl