AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #385: AlDente not working on M1 11.6?

scoopdreams, 2021-09-18:
After upgrading to MacOS 11.6, AlDente seems to have stopped working.

Have tried reinstalling the helper, removing AlDente and reinstalling it entirely, etc. Nothing seems to work.

Could 11.6 have borked AlDente?

[image]

nguye205, 2021-09-19:
mine is working fine except for sailing mode.

scoopdreams, 2021-09-19:
> mine is working fine except for sailing mode.

Thanks, so this is on M1 Mac and you just disabled sailing mode for it to operate?

nguye205, 2021-09-19:
> > mine is working fine except for sailing mode.
> 
> Thanks, so this is on M1 Mac and you just disabled sailing mode for it to operate?

Yes, it is on M1 mac. It does stop charging at the set limit but the sailing mode doesn't work. I set the charging limit to 70% and the sailing mode limit set to 35%. So in theory, it shouldn't charge until the battery go down to 35% but it still charges to 70% when the battery is above 35%.

simotau, 2021-09-21:
Is it me, or the developers are not replying to the issues opened anymore?

dragomirweb, 2021-09-30:
I can confirm, sailing mode not working and sometimes it charges over the limit 2% 5% 10%. Are the developers aware of this issues?

MatthiasKerbl, 2021-10-01:
> Yes, it is on M1 mac. It does stop charging at the set limit but the sailing mode doesn't work. I set the charging limit to 70% and the sailing mode limit set to 35%. So in theory, it shouldn't [...]

Hi @nguye205,

You can read more about the functionality of the Sailing Mode in on our website: https://apphousekitchen.com/feature-explanation-sailing-mode/

Cheers,
Matthias

MatthiasKerbl, 2021-10-01:
> After upgrading to MacOS 11.6, AlDente seems to have stopped working.
> 
> Have tried reinstalling the helper, removing AlDente and reinstalling it entirely, etc. Nothing seems to work.
> 
> Could 11.6 have borked AlDente?
> 
> [image]

Hi @scoopdreams,

Do you still experience this issue?
Have you tried restarting/reinstalling AlDente? Sometimes after bigger macOS updates this needs to be done.

Cheers,
Matthias

scoopdreams, 2021-10-03:
Hi Mathias, tried reinstalling AlDente entirely again, and curiously it seems to have worked.

Will mark this issue as closed, thanks!

closed 2021-10-03 by scoopdreams