AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #387: Macbook Pro automatically shuts down in Calibration Mode

FaeAuditore, 2021-09-23:
There is something wrong with the battery health of my MacBook Pro (15-inch, 2016, Big Sur 11.5.2) . Sometimes even if the value on battery is around 25%, Mac will automatically shuts down. Then I tried the Calibration Mode in AlDente Pro, my Mac still shuts down in the process of discharging. It seems that the Calibration Mode cannot complete on my Mac. Is there any solution? Thanks.

MatthiasKerbl, 2021-09-25:
Hi @FaeAuditore,

it sounds like your battery calibration is disturbed.
Try to manually discharge and charge your battery multiple times until you can discharge your MacBook to 0% again.

Cheers,
Matthias

FaeAuditore, 2021-09-27:
Thank you very much,  @MatthiasKerbl . I will have a try:)

cheack, 2021-10-19:
Hi @MatthiasKerbl,
How many times do I need to do a discharge/recharge cycle? I've already done it five or six times, but my mac still shuts down at 50 percent. After charging to 100%, I leave it on charge for 2-3 hours, then unplug it, but the charge shows 100% for 1-2 hours, only after that it starts to decrease.

MatthiasKerbl, 2021-10-21:
> Hi @MatthiasKerbl, How many times do I need to do a discharge/charge cycle? I've already done it five or six times, but my mac still shuts down at 50%. After charging to 100%, I leave it on charge for 1-2 hours, then unplug it, but the charge shows 100% for 1-2 hours, only after that it starts to decrease.
> 
> @FaeAuditore, were you able to discharge your battery?

Hi @cheack,

It differs from MacBook to MacBook but it if did not get better after discharging and charging 10 times, the problem might not be a miscalibrated battery management system but an aged or defective battery instead.

Cheers,
Matthias

ReadyCloudSuite, 2021-10-22:
@MatthiasKerbl, I tried several more times to discharge the macbook, but without success. Then I left it on the charger for the whole day (8 hours), and only after that it started discharging immediately after taking it off the charger. Perhaps all of this happened because I hadn't calibrated my battery in almost a month. 

By the way, what's the best way to calibrate the battery - charge to 100, then discharge to 0 and immediately set the limit? Or discharge to 0, then back to 100, then back to 0 and only then set the limit?
Thanks.

cheack, 2021-10-22:
@MatthiasKerbl, I tried several more times to discharge the macbook, but without success. Then I left it on the charger for the whole day (8 hours), and only after that it started discharging immediately after taking it off the charger. Perhaps all of this happened because I hadn't calibrated my battery in almost a month. 

By the way, what's the best way to calibrate the battery - charge to 100, then discharge to 0 and immediately set the limit? Or discharge to 0, then back to 100, then back to 0 and only then set the limit?
Thanks.

closed 2021-12-02 by MatthiasKerbl