AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #389: AlDente Pro on MacBook 16'' 2021

lboisset, 2021-09-23:
I'm having a very strange performing on the battery. It doesn't charge, even if I switch off Sailing mode. I love the idea behind AlDente because I use a Dock to connect multiple monitors which already charge the Mac, so I have no option to disconnect it, but let's say tomorrow I'm gonna move from my office and I need to ensure my Mac is on 100% battery. I can't get it with aldente installed, no matter what I change on its configuration.

MatthiasKerbl, 2021-09-25:
> I'm having a very strange performing on the battery. It doesn't charge, even if I switch off Sailing mode. I love the idea behind AlDente because I use a Dock to connect multiple monitors which [...]

Hi @lboisset,

do I understand you correctly that when your MacBook is plugged in and you set the charging limit to 100% with AlDente, your MacBook does not charge to 100% even after an hour?

Cheers,
Matthias

lboisset, 2021-09-25:
Hi Matthias,

Exactly. Not only an hour, it's been like this for two days. Even shutting down during the night the battery keeps going down up to 6% and not charging, but Sailing mode was off since the day before. There I decided to uninstall because I really need a full charge. But it doesn't charge either, no matter I restart a couple of times. I force a refresh pressing the power on button for 15 secs to recover normal behavior. 

I didn't install it jet waiting for any help.

Cheers,

Luis

MatthiasKerbl, 2021-09-26:
> Hi Matthias,
>
> Exactly. Not only an hour, it's been like this for two days. Even shutting down during the night the battery keeps going down up to 6% and not charging, but Sailing mode was off since [...]

Did you ever use the Intel Mode of AlDente Pro?
Did you try to do an SMC reset? (https://support.apple.com/en-us/HT201295)

Cheers,
Matthias

lboisset, 2021-09-26:
No, I didn't try with Intel Mode. I don't have a clear idea of the differences. I see that it will only have the charge limit capability while what I really want it to been able to discharge the battery while I'm still connected to the dock for external monitors. This is something I was able to do with my 2014 MacBook and Thunderbolt2 dock but not with the new usb-c docks.

Did you mean desert SMC while still using AlDente to see if it's behavior change? No. I only did it after remove it because charge limits stills remains set. Even I set limit to 100% before uninstalling. That happens to me as well in my mid2014 MacBook Pro (yes AlDente works there as well). 

So you propose me to install it again, reset the SMC and check how it goes?

Regards

Luis

MatthiasKerbl, 2021-09-26:
> No, I didn't try with Intel Mode. I don't have a clear idea of the differences. I see that it will only have the charge limit capability while what I really want it to been able to discharge the [...]

The Intel Mode uses a different pathway to control the charging behavior of MacBook which is mostly intended for pre Thunderbolt 3 MacBooks since they don’t have the new technical pathways available. That’s why you had to use the Intel Mode on your 2014 MacBook.

No, I meant doing an SMC reset when AlDente Pro is uninstalled. Please try to reinstall the newest version of AlDente Pro and if it might work again. If it doesn’t work, please send us a debug file. Preferably on our premium support email address which you can find in the release notes in Setapp.

Cheers,
Matthias

MatthiasKerbl, 2021-11-15:
Hi @lboisset,

Do you still encounter this issue?

Otherwise, I will close this issue for now.

Cheers,
Matthias

lboisset, 2021-11-15:
Hi Matthias, close it.

I’m still having weekly strange behavior (batt going to 5% and not loading from thundertbolt dock, but yes from power unit) but I didn’t have time to make a detailed verification if it’s related to AlDente. So I’ll be open a new one if I could verify it is for you.

thanks

closed 2021-11-15 by MatthiasKerbl