AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #391: Docking station interference

elearning-x, 2021-09-24:
Hello 

I'm using MacbookPro M1 2020 os 11.6 with a docking station Urban Factory and AlDente Free 1.07.
The charger is plugged on the docking station.

With this configuration AlDente doesn't limit the charge to 80%.

Is this normal ?

Oll0w4in, 2021-09-24:
No but one problem could be, that you have not discharched enough before setting the limit and plug in again. 
in my case i had to discharge to 30% until my limitation from 55% was working. 
so if you did not discharged a bit try this.

MatthiasKerbl, 2021-11-05:
> Hello
> 
> I'm using MacbookPro M1 2020 os 11.6 with a docking station Urban Factory and AlDente Free 1.07. The charger is plugged on the docking station.
> 
> With this configuration AlDente doesn't limit the charge to 80%.
> 
> Is this normal ?

Hi @elearning-x,

Usually, it does not matter if your MacBook is connected to a power brick or a docking station.
Do you still encounter this problem?
Could it be possible that your MacBook was in sleep while connected to the docking station?

Cheers,
Matthias

closed 2021-11-28 by MatthiasKerbl