AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #397: Battery not charging at all after uninstall

backhand, 2021-10-06:
Device: 2020 13" Macbook Pro, MacOS 11.6

I installed AlDente to manually limit my charge to 80%, but it seemed to create some weird lockups when plugging in or out a charging USB-C cable, so I uninstalled the helper app and quit the main app. Now the battery simply doesn't charge at all. I've tried disabling "Optimized battery charging" and Manage battery longevity".
The only thing that helped was reinstalling Aldente with the helper app, but if I uninstall it again the charging stops. Any suggestions on how to revert this behaviour? Let me know if you need more details too.

MatthiasKerbl, 2021-11-15:
Hi @backhand,

Do you still encounter this issue?

Otherwise, I would close this issue for now.

Cheers,
Matthias

closed 2021-12-02 by MatthiasKerbl

itsJosephV, 2022-10-07:
the first time I deleted Aldente I wasn't aware of the steps te be taken to uninstall it, now whenever I plug it no matter the battery level,  it shows me the plug icon like if "charging was on hold", it's like some kind go bug from this App impregnated in my macbook system, I've tried  everything, from reinstalling the app and follow the steps to delete it properly, also manually, even reinstalled the OS but nothing seems to help, why is this App so damn intrusive?, how do I fix this? my macbook was working flawless before this App. it's a m1pro btw, is it even optimised for this model?

EnigmaticZee, 2023-02-22:
> the first time I deleted Aldente I wasn't aware of the steps te be taken to uninstall it, now whenever I plug it no matter the battery level, it shows me the plug icon like if "charging was on hold", it's like some kind of bug from this App impregnated in my macbook system, I've tried everything, from reinstalling the app and follow the steps to delete it properly, also manually, even reinstalled the OS but nothing seems to help, why is this App so damn intrusive?, how do I fix this? my macbook was working flawless before this App. it's a m1pro btw, is it even optimised for this model?

Were you able to solve the issue?

hansolowireless, 2024-05-26:
> Were you able to solve the issue?

Did you solve this?