AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #408: AL Dente seems not limiting charging on M1 mac

nanoporous, 2021-10-19:
tried reinstalling and restarting several times but problem persist. Strangely enough, I can activate discharge with Al Dente.
Any hints?

nanoporous, 2021-10-21:
> Hi @nanoporous <https://github.com/nanoporous>,
>
> Please describe your issue in more detail.
> Do you mean that AlDente does not limit charging at all on your device or
> only in certain situations?
>
> Cheers,
> Matthias

Hi, does not limit charging at all

MatthiasKerbl, 2021-11-02:
Hi @nanoporous,

Did you try our newest version AlDente 1.08 yet?
Do you still encounter this issue with the newest version?

Cheers,
Matthias

nanoporous, 2021-11-02:
Yes, it is working.

closed 2021-11-02 by MatthiasKerbl

nanoporous, 2021-11-02:
i am sorry for reporting it's not limiting charing again. Seems by letting
it discharge would help

nanoporous, 2021-11-03:
ACtually i think it got charged to 100% while mac was asleep.
When waken up the software was not working, but it works after logging out
and logging in

MatthiasKerbl, 2021-11-03:
Hi @nanoporous,

AlDentes capabilities of limit charging are highly limited while your MacBook is in sleep or powered off.
You can read more about this behavior and what you can do about this on our blog:

- https://apphousekitchen.com/feature-explanation-stop-charging-when-powered-off-app-closed/
- https://apphousekitchen.com/feature-explanation-stop-charging-when-sleeping/

Hope that helps.
Cheers,
Matthias

nanoporous, 2021-11-04:
thanks! That's very helpful.
Idont't get notifications from al dente that asks me to turn off "optimized
charging" in mac settings, will it interfere Al dente at all?

MatthiasKerbl, 2021-11-04:
> thanks! That's very helpful. Idont't get notifications from al dente that asks me to turn off "optimized charging" in mac settings, will it interfere Al dente at all? Matthias Kerbl ***@***.***> 於 2021年11月3日 週三 上午2:02寫道：
> […](#)

Yes, you definitely have to deactivate "Optimized battery charging". Otherwise, your MacBook will keep charging when the charge limit is reached.

nanoporous, 2021-11-09:
may I get a license of pro for more evaluation? I could provide more
feedback if needed

MatthiasKerbl, 2021-11-09:
> […](#)

Hi @nanoporous,

If you are interested in AlDente Pro, you can buy it here on our website:
https://apphousekitchen.com/pricing/

Does AlDente still not limit charging? You could send us a debug file.

Cheers,
Matthias