AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #423: Mabook Air 2020 M1 charging limiter goes past the percent I set.

dbz2k, 2021-10-29:
Hello for some example say I set the charge limit to 86 it eventually goes past this percentage in a couple days or so. AlDente Classic 2.2 does not have this issue

MatthiasKerbl, 2021-10-30:
> for some example say I set the charge limit to 86 it eventually goes past this percentage in a couple days or so. I did notice it increases in sleep modes and I never saw it increase while I was using it. AlDente Classic 2.2 does not have this issue.
> 
> I even ran the Apple Diagnostics hardware test to see if something was wrong with my battery or something, and it showed no issues found.
> 
> I'm on Big Sur 11.6

Hi @dbz2k,

Unfortunately, AlDente can't control the charging behavior on your MacBook when it is in sleep. Therefore it continues charging when you shut it down or when it goes into sleep. However, AlDente Pro has a feature called "Stop charging when in sleep" which will stop charging right before it goes into sleep and therefore, it will not charge to 100%.

Cheers,
Matthias

dbz2k, 2021-11-01:
I just bought the pro upgrade, and so does that mean it will not charge at all during sleep mode?

MatthiasKerbl, 2021-11-01:
> I just bought the pro upgrade, and so does that mean it will not charge at all during sleep mode like it will directly be connected by power from the outlet, and will not charge to my set limit during sleep mode?

Yes, since AlDente can't do anything while your MacBook is in sleep, it can't stop charging at the set percentage. Therefore, with the feature "Stop charging when in sleep", AlDente will stop charging right before your MacBook goes into sleep and your MacBook will stay at the battery percentage it had when it went into sleep.

Cheers,
Matthias

MatthiasKerbl, 2021-11-05:
Hi @dbz2k,

We have released new articles with detailed explanations about those features on our blog:

- https://apphousekitchen.com/feature-explanation-stop-charging-when-sleeping/
- https://apphousekitchen.com/feature-explanation-stop-charging-when-powered-off-app-closed/

Cheers,
Matthias

closed 2021-11-05 by MatthiasKerbl