AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #424: MBP 2017 Intel - Several bugs in 1.08 and Monterey

tosbsas, 2021-10-30:
Hi there

I do see behavior of Al Dente 1.08 in Monterey I didn't see in 1.07 and BigSur

1. Battery is set to max 80% but I get always mor than 80, floating is set to 5%
2. After using topUp machine charged to 100% instead of 80% it was set to

Please fix

MatthiasKerbl, 2021-10-31:
> Hi there
> 
> I do see behavior of Al DentePro 1.08 in Monterey I didn't see in 1.07 and BigSur
> 
> 1. Battery is set to max 80% but I get always mor than 80, floating is set to 5%
> 2. After using topUp machine charged to 100% instead of 80% it was set to
> 
> Please fix

Hi @tosbsas,

1. I am currently not experiencing any problems in this regard with my MBP 2016 Intel. In which situations exactly does your MacBook charge above the Charge Limit?
2. That’s definitely a bug. Thanks for pointing it out, we will fix it with the next update!

Cheers,
Matthias

tosbsas, 2021-11-01:
1. hm, I do See values higher than 78 with 78 set as limit. Normal use

2. good to know that its really a bug - hoping for the fix

closed 2021-11-01 by tosbsas