AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #425: Pressing enter in text fields changes their value.

letheed, 2021-10-30:
If I click in one of the text fields and press enter, the value changes.

In the following video I first click in the Heat Protection text field, then press enter. The value is set to 0 instead of the already present value. I then type 40, press enter again, and the value is set properly.
The same behaviour happens with the Charge Limit and the Sailing Mode text fields.

https://user-images.githubusercontent.com/5330692/139557160-6c78c90b-7a24-409e-9291-023f666a0e1c.mov

MatthiasKerbl, 2021-10-31:
Hi @letheed,

I can imagine that it has something to do with AlDente not recognizing yet that you updated to macOS 12.
Did you try to restart AlDente and your MacBook?
You could also try to reinstall AlDente.

Cheers,
Matthias

dionvl, 2021-11-01:
I have the same issue. MBP M1, macOS 12.0.1. It even messes with charging behavior, I have set the charge limit to 80%, but sailing mode thinks the limit is at 20% when I first click on it? 

https://user-images.githubusercontent.com/16754743/139649823-9b0f22d0-57fe-4efd-929a-3020b6ba3d4e.mov

MatthiasKerbl, 2021-11-01:
> I have the same issue. MBP M1, macOS 12.0.1. It even messes with charging behavior, I have set the charge limit to 80%, but sailing mode thinks the limit is at 20% when I first click on it?
> 
>  Schermopname.2021-11-01.om.10.16.41.mov

Hi @dionvl,

Did you try my suggestions from earlier?
"Did you try to restart AlDente and your MacBook?
You could also try to reinstall AlDente."

Cheers,
Matthias

dionvl, 2021-11-01:
Yes I have rebooted my MacBook and did a clean reinstall of AlDente, the issue unfortunately persists. I think all text fields are glitched the first time you interact with them in between app launches. I'm not sure if this is merely a visual issue, or if charging behavior is also affected.

letheed, 2021-11-01:
@MatthiasKerbl I have reinstalled AlDente and restarted my MacBook several times. I can confirm that the behavior is very consistent.

MatthiasKerbl, 2021-11-04:
Hi Everyone,

We have just released version 1.09 of AlDente which has a fix for the bug you encountered.
https://github.com/davidwernhart/AlDente/releases

Cheers,
Matthias

closed 2021-11-04 by MatthiasKerbl