AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #426: Limit Charing RATE?

Snipeye, 2021-10-30:
Hello!  Electrical engineer here.

I know that the voltage to which you charge your batteries (and to which you discharge them) impacts their longevity.  However, with my background, I also know that the rate at which you charge them makes an impact - that's often related to the temperature at which they charge.

I see that Al Dente has an option to control max charge temperature, and I think that's fantastic!  I'm considering buying a pro license, but I'm curious if there are any plans (or, indeed, if there is any ability in the system) to limit the _rate_ at which the batteries charge, rather than just the max percent.

Thanks!

MatthiasKerbl, 2021-10-31:
> Hello! Electrical engineer here.
>
> I know that the voltage to which you charge your batteries (and to which you discharge them) impacts their longevity. However, with my background, I also know that [...]

Hi @Snipeye,

Changing the charging rate is definitely something we would love to do. However, we have not found a pathway to do that and we are not sure if it is even possible internally.

The other option of course is to use a less powerful power brick. I use a 30W charger instead of the 87W for my 15" MacBook and it works fine most of the time.

Cheers,
Matthias

PearApple151, 2021-10-31:
> Hi @Snipeye,
>
> Changing the charging rate is definitely something we would love to do. However, we have not found a pathway to do that and we are not sure if it is even possible internally.
>
> The [...]

I thought about that, but I read in someplaces (not reliable places, but I hate to take chances with expensive equipment) that using underpowered charges could damage your computer. Cause shorts or melt something on the board or something. I don't remember what I read, and it wasn't from a reliable source anyway (probably Reddit or Quora) so I hate to waste too much time trying to find it. You obviously disagree, could explain the misconception there? 

I saw one person saying that it doesn't matter what wattage your brick is as long as the voltage is correct. if the voltage is correct and it is underpowered, the worst that will happen is that your computer will discharge. and maybe the brick will overheat???. 

anyways, all of the random muses of internet people who may or may not know what they're talking about aside. what's the straight dope on using bricks that are over and underpowered.

Pixel3D, 2021-10-31:
Just buy quality chargers and cables. The MacBook decides the maximum power it is getting from the charger so a larger charger is fine. If it is more than the charger can deliver then the MacBook will charge slower. The charger won't deliver more power than what it is made to do. So I wouldn't say they overheat but with more maximum usage of the charger the lifespan could be shorter.

It wouldn't be practical, but you could use a 5W charger to charge the MacBook over night. Even if the MacBook came with a 96W charger you could also use a 140W charger without problems.

Snipeye, 2021-11-01:
Thanks for the input, everybody - @MatthiasKerbl I figured the OS might not actually have an API to hook into for this (given the seeming difficulty with getting a fixed battery % cutoff for charging), but the recommendation to just use a lower-powered charger is sensible.  I'm going to do some sniffing on the data passed through magsafe and see if I can get my charger to report as a lower wattage than it actually is; I wonder if that would prompt a slower charge rate.

DamonHD, 2021-11-07:
I would very much be on for this, eg to limit peak power demand shen charging off grid, or when on grid but the grid is having a bad-hair moment!

Thanks for making this work at all!

Rgds

Damon

PS. A write-up I did on this sort of thing... http://www.earth.org.uk/Hey-Siri-Help-Flatten-The-Duck.html

xawill, 2021-11-16:
Also interested by this. I have a brand new MacBook Pro 16" M1 Pro which comes with the 140W charger. I don't need to charge the battery at that rate, so I currently use the original Apple charger, but with a USB-C charging cable (limited at 100W by the Power Delivery protocol) instead of the USB-C to MagSafe 3 cable. It is a workaround, but it would be great to be able to use the MagSafe connector at 100W.

KonradDIKU, 2021-12-02:
From the duplicate that I created: @MatthiasKerbl - "However, we have not found a technical pathway to do it yet and we do not know if it is even possible."

Perhaps it could be accomplished by altering the SMC settings? For a reference you could take a look on how it can be done with a fan speed settings: https://github.com/hholtmann/smcFanControl/blob/master/smc-command/smc.c

It might involve some reverse engineering as I do not know if Apple has released their docs describing SMC. Furthermore, even if the wattage can not be altered there(which I doubt) you can always fake the battery temp sensor readings and relay on macOS* to trim down charging rate automatically while still presenting the correct values in the app itself ;) 

* https://support.apple.com/en-us/HT201678

Snipeye, 2021-12-23:
Is AlDente applescript-compatible?

I've done some sniffing, and it seems like when you're using MagSafe (or, obviously, USB-C) the USB-PD spec is used to report charging allotment.  A drop-in device between the charger and the charging cable could modify the PD packets to report a lower maximum (and therefore, presumably, the mac would REQUEST that maximum) resulting in a lower charge rate.

Based off what I know about the USB-PD spec, it goes something like: cables are plugged and discovery begins.  Some packets are exchanged and one device decides it will be the host and the other will be the client (for power).  Client requests more power from the host, host responds with the power profiles it can handle.  Client selects a profile and requests it, host switches voltage and begins supplying.

It's not CONCRETE evidence, but based on the fact that apple didn't release a magsafe 2 to magsafe 3 adaptor, I'm hypothesizing that they may have changed the data scheme they used.  Typically magsafe has 2x GND (outermost, IIRC), 2x V+ (next pins in), and then a central data tap.  Might have that mixed up.  Anyway, deciding charging rate is done with a 1-wire protocol on the data tap through a proprietary data scheme.  At least, that WAS the story for magsafe 1 and 2.  I think magsafe 3 might just plain use the USB-PD spec now (or "EPD" spec which supports up to 240W), which might explain why apple didn't release a 2->3 adaptor for magsafe; it would be incompatible from a protocol standpoint.  Not impossible to fix, but maybe more hassle than apple wanted to deal with?

Anyway, the upshot of all this: at some point, the macbook is requesting - either POSSIBLY through magsafe, but definitely through the USB port when a USB port is used for charging - some power profile from the charger.  If we can find when that's happening and request a different power profile (based on what the charger reported as available) we can make the CHARGER provide a lower wattage, which should accomplish the goal.

As an aside - and why I asked if AlDente supports applescript - you could maybe make a semi-ghetto workaround:
When a charger is plugged in, AlDente uses the system information to find what wattage it is.  If it's higher than we want, we tell AlDente to set the max charge limit to whatever the current battery percentage is.  That'd stop charging for the time being.  If we have a 100W charger and we effectively want to charge at 30W, then we set the max charge % higher (back to whatever we were using) for 30 seconds or something, then lock it again at current battery % for another 70 seconds.  I don't know how fine-grained you could get, and it's far from ideal, but it's better than nothing.  I ask about applescript because most of this could be done with applescript provided AlDente can report "is charging", "current battery %", and "charger wattage" and accept commands to change "max charge %".

I'm rambling at this point.  Anyway, thanks again for the continued development, my last MBP (2015) died because the battery was just fried, so I'm keeping my 2021 between 25-75% and it still lasts basically all day, and I know I'm keeping the battery super healthy.  Feels good.

TimBeyer, 2024-04-10:
I have an external monitor supporting PD that due to some design flaw turns off once a laptop actually draws the 94W it advertises. Apparently it cannot actually handle delivering as much power as it says it can deliver. A feature like described here would be so helpful in restoring it to being useful since the manufacturer doesn't seem to be able to fix it.
Currently I always have to go and top up my laptop to close to 100% because then it won't charge at full power and the screen won't crash.

For me this also seems to be a good indicator that something like this should be possible to control in software.
A mac won't simply charge at 140W or whatever power adapter you have connected to it until it's full.
It will lower the power the closer it gets to 100%.

You can also see that here:
[image]

Of course this could have been implemented somewhere on the very low levels of the device, but I'm still hopeful that one day this can be fixed in software.