AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #427: Will be 100% charged after two days of standby

wyfang, 2021-11-01:
I keep my laptop in the office and if I don't work on Saturday and Sunday, I turn on my computer every Monday and it will be filled to 100%
[image]
[image]
[image]

MatthiasKerbl, 2021-11-01:
Hi @wyfang,

Which MacBook model do you have?

Cheers,
Matthias

wyfang, 2021-11-01:
> Hi @wyfang,
> 
> Which MacBook model do you have?
> 
> Cheers, Matthias

macbook pro 2016
[image]

wyfang, 2021-11-01:
[image]
It looks like I need to reset the click to 55%.

MatthiasKerbl, 2021-11-01:
The problem is that your MacBook is in sleep or powered off during the weekend and AlDente can't control the charging behavior while your MacBook is in sleep or powered off.
You should activate the features "Stop charging when sleeping" and "Stop charging when powered off". With those features activated, AlDente will stop charging right before your MacBook goes into sleep or before it gets powered off. Therefore, your MacBook will stay at the charge level it had right before it went into sleep or before it got powered off.

Cheers,
Matthias

closed 2021-11-01 by MatthiasKerbl

wyfang, 2021-11-03:
> The problem is that your MacBook is in sleep or powered off during the weekend and AlDente can't control the charging behavior while your MacBook is in sleep or powered off. You should activate the [...]

I tried activating "stop charging when sleeping" and "stop charging when switching off" and a day has passed and the battery still has 2% left. I have absolutely no idea what to do. When I switch it on, the battery becomes 2% after one day, when I switch it off, it becomes 100% after two days.
[image]
[image]
[image]

MatthiasKerbl, 2021-11-03:
Hi @wyfang,

AlDente will not charge your MacBook when it is in sleep or powered off. Instead, it will just hold the current charge limit when the features "Stop charging when sleeping" and "Stop charging when powered off" are activated.

Please read our detailed explanation about those two features on our blog:

- https://apphousekitchen.com/feature-explanation-stop-charging-when-powered-off-app-closed/
- https://apphousekitchen.com/feature-explanation-stop-charging-when-sleeping/

Cheers,
Matthias