AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #431: Battery power drop seriously

leechaojie, 2021-11-03:
Battery power drop serious, 9 p.m. charge to 60%, close the lid, the morning power down to 23%，
MacBook 16 2019
AlDente Free 1.08
macOS Monterey v12.0.1

and  another MacBook pro 13 2019 4 Thunderbolt 3
AlDente Free 1.08
macOS Big Sur v11.4

MacBook pro 13 is the AlDente Free 1.08 that I installed yesterday to see the battery consumption of the box cover hibernation, and the result is the same, the battery is down to 36%.

abeldemoz, 2021-11-03:
I have the same issue.

leechaojie, 2021-11-04:
I tested it on purpose yesterday. When I quit AlDente at night before going to bed, at this time, the power was 80%, and when I opened the MacBook at 9 am, the power dropped to 76%, which is a reasonable range.

MatthiasKerbl, 2021-11-04:
Hi @leechaojie and @abeldemoz,

Thank you for reporting this issue.
We are currently looking into this and will fix it as soon as possible.

Cheers,
Matthias

MatthiasKerbl, 2021-11-15:
Hi @leechaojie and @abeldemoz,

We have just released a new version of AlDente (1.10). This version should fix the issues you have experienced with AlDente.
https://github.com/davidwernhart/AlDente/releases

Thank you for your patience!

Cheers,
Matthias

leechaojie, 2021-11-17:
I'm sorry, I tested AlDente (1.10) on my MacBook the day before yesterday and the battery of my MacBook still dropped from 60% to 37% after one night, after I switched back to version 2.2, the battery level is normal this morning, my other MacBook pro 13 has the same problem.
I downloaded AlDente (1.10) directly and replaced it within the application, and AlDente (2.2) replaced 1.10 in the same way, and that's all I did.
abeldemoz <https://github.com/abeldemoz>,your problem is solved Did you solve your problem?

MatthiasKerbl, 2021-11-17:
Hi @leechaojie,

Thank you for reporting back and I am sorry that you still have issues with AlDente.
There might be an issue with just replacing the program files of AlDente Classic 2.2 and AlDente Free 1.10 since the helpers will not be replaced like that.
Please try to completely reinstall AlDente Free 1.10.
Follow the steps in this uninstallation guide (this way, the helpers will be removed too): https://apphousekitchen.com/how-to-uninstall-aldente-pro/
Download AlDente Free 1.10 here and install it again: https://github.com/davidwernhart/AlDente/releases

PearApple151, 2021-11-19:
> I'm sorry, I tested AlDente (1.10) on my MacBook the day before yesterday and the battery of my MacBook still dropped from 60% to 37% after one night, after I switched back to version 2.2, the [...]

what exactly is happening. before you go to bed you say you quit al dente? Like you close the program completely?

leechaojie, 2021-11-19:
I re-tested it last night, and since I installed version 2.2, which did not offer the option to remove Helper, I uninstalled it with ClearMyMac, restarted the computer, installed version 1.10, then removed Helper, exited AlDente and removed it in Finder, restarted the MacBook again, and reinstalled AlDente ( 1.10). The battery power limit is 100%.
When I turned on the computer this morning, the battery level dropped to 83%.
Is my MacBook the only one having this problem? But I have this problem with my other MacBook too. It's interesting and strange.

leechaojie, 2021-11-19:
> > I'm sorry, I tested AlDente (1.10) on my MacBook the day before yesterday and the battery of my MacBook still dropped from 60% to 37% after one night, after I switched back to version 2.2, the battery level is normal this morning, my other MacBook pro 13 has the same problem. I downloaded AlDente (1.10) directly and replaced it within the application, and AlDente (2.2) replaced 1.10 in the same way, and that's all I did. abeldemoz https://github.com/abeldemoz,your problem is solved Did you solve your problem?
> > […](#)
> 
> what exactly is happening. before you go to bed you say you quit al dente? Like you close the program completely?

No, I just closed the lid of my MacBook after I got off work and did nothing, nothing quit.

leechaojie, 2021-12-20:
Wow, good news, problem solved!
When I uninstalled version 2.2 and installed 1.11, the battery consumption during sleep stabilized, but not immediately, the drop was still severe the first night, I uninstalled the software again, reset the SMC, installed it again, and the power consumption stabilized the second night, the above steps did not work with 1.08 and 1.10, thank God 1.11 fixed it.
Actually there was another long term problem that was also fixed, after I installed and uninstalled 1.10 on my other MacBook, the battery consumption in sleep state was still severe, resetting SMC didn't work either, I believe I uninstalled it correctly and I had to turn off my computer first every night as a way to avoid it consuming too much power.
Both MacBooks are fixed, but I have a question, what is the cause of the abnormal battery consumption in sleep state? I can't find the power-hungry app on Activity Monitor either.

MatthiasKerbl, 2021-12-20:
Hi @leechaojie,

I am glad your issue is finally fixed. 
So far we were not able to reproduce your bug. However, we made a lot of improvements regarding the prevent sleep setting which might have solved it.
In general, I do not think your issue is that AlDente or another app consumes too much power, rather than your MacBook not going into sleep at all.

I will close this issue now but feel free to report back if you find anything else.

Cheers,
Matthias

closed 2021-12-20 by MatthiasKerbl