AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #432: Al-Dente Pro Calibration Bug

abeldemoz, 2021-11-03:
Al-Dente Pro user here. My battery was at 8% and charging. I turned on calibration and the battery started discharging. It shouldn't start discharging unless it's over 15%. Please fix.

MatthiasKerbl, 2021-11-03:
Hi @abeldemoz,

I just tried it and you are right, AlDente starts discharging but only for half a second and immediately starts charging to 100%.
Is it the same for you?

Cheers,
Matthias

abeldemoz, 2021-11-04:
> Hi @abeldemoz,
> 
> I just tried it and you are right, AlDente starts discharging but only for half a second and immediately starts charging to 100%. Is it the same for you?
> 
> Cheers, Matthias

No, it just kept discharging. I had to stop calibration by the time the battery discharged to 1% and wait for the battery to charge to 15% before starting calibration again.

Also, AlDente-Pro is causing my battery to drain overnight. This was my battery last night. On other occasions, I quit AlDente-Pro before going to sleep and my battery doesn't drain. This is a critical bug that needs to be fixed for two reasons:
1. The purpose of AlDente-Pro is to save battery by preserving health
2. As a user, I shouldn't have to remember to quit AlDente-Pro every time my MacBook Pro isn't connected to power and open it again when connected to power.

I'd expect this from free software, not paid software. Please fix the battery drain issue.
[image]

MatthiasKerbl, 2021-11-04:
> No, it just kept discharging. I had to stop calibration by the time the battery discharged to 1% and wait for the battery to charge to 15% before starting calibration again.

Interesting that you encounter a different behavior of the Calibration Mode than I do.
Which MacBook model do you have?
Which macOS version do you have installed?
Do you use the newest version of AlDente (1.08)?

Thank you for reporting the battery drain issue, we will fix it as soon as possible.

Cheers,
Matthias

abeldemoz, 2021-11-04:
> Interesting that you encounter a different behavior of the Calibration Mode than I do. Which MacBook model do you have? Which macOS version do you have installed? Do you use the newest version of [...]

2020 13" MacBook Pro i5 2.0GHz 16GB RAM 512GB SSD
macOS Monterey (12.0.1)
AlDente-Pro v1.08

lbydev, 2021-11-16:
I am having a problem that when I re-plug the power adapter, it does not continue to auto-calibrate。

MatthiasKerbl, 2021-11-16:
> I am having a problem that when I re-plug the power adapter, it does not continue to auto-calibrate。
> 
> macOS Monterey (12.0.1) AlDente-Pro v1.09

Hi @lbydev,

Currently, you have to keep your MacBook plugged in from start to finish of the Calibration Mode.

Cheers,
Matthias

Lennix, 2021-12-02:
I also plugged in below 15 percent, it started charging and then I was able to click on "Start calibration" (would be great if I could enable it before plugging in - it detects that I'm already discharged below 15% and begins with the next step).
I did not charge immediately. I believe it started once it dropped another percent.

MatthiasKerbl, 2021-12-02:
> I also plugged in below 15 percent, it started charging and then I was able to click on "Start calibration" (would be great if I could enable it before plugging in - it detects that I'm already [...]

Hi @Lennix,

Thanks for reporting the behavior of the Calibration Mode on your MacBook. It seems like it is exactly the same as on my MacBook. We will improve the algorithm of the Calibration Mode in a future update.

Cheers,
Matthias

MatthiasKerbl, 2022-01-16:
Hi there,

We have just released a new version of AlDente the completely refreshed Calibration Mode which should fix the issues explained here.
You can download the newest version [here](https://github.com/davidwernhart/AlDente/releases) or just do the update within the app.

Cheers,
Matthias

closed 2022-01-16 by MatthiasKerbl