AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #439: Charges Past Specified Level

abeldemoz, 2021-11-05:
I have a 13-inch 2020 MacBook Pro. When I use the 61W power adapter, everything is fine. When I use a 96W power adapter, it charges past the limit I specified. I set the limit to 80% when connected to the 96W adapter and it charged to 91%.

MatthiasKerbl, 2021-11-06:
Hi @abeldemoz,

Are you certain that you get different behavior with different power adapters? Or could it be that your MacBook went into sleep or got powered off while you used it?
Since AlDente's capabilities of controlling charging while your MacBook is in sleep or powered off is highly limited.
You can read more about this topic on our blog:

- https://apphousekitchen.com/feature-explanation-stop-charging-when-sleeping/
- https://apphousekitchen.com/feature-explanation-stop-charging-when-powered-off-app-closed/

I have always different charging adapters with different charging powers in use and have never experienced issues with AlDente while using different charging adapters.

Cheers,
Matthias

abeldemoz, 2021-11-06:
This behaviour occurs even when the MacBook Pro is awake and in use. The 61W doesn’t charge past the charging limit, whereas the 96W frequently charges past the charging limit.

MatthiasKerbl, 2021-11-06:
> This behaviour occurs even when the MacBook Pro is awake and in use. The 61W doesn’t charge past the charging limit, whereas the 96W frequently charges past the charging limit.

Do you have an Intel or Apple Silicon M1 MacBook?

abeldemoz, 2021-11-06:
Intel MacBook Pro 13" 2020 i5 2.0GHz 4-core 16GB RAM 512GB SSD

MatthiasKerbl, 2021-11-07:
Hi @abeldemoz,

Unfortunately, I am currently not able to reproduce this issue. I have tried it on multiple Intel-based MacBooks with various power adapters and charging past the charge limit never occurred.
Do you have any recommendations on how I can reproduce this bug?

tosbsas, 2021-11-08:
I do See the Same but Not this far up. Set to 80 y get 81 or 82 After  several days being connected. 

Its set to 80 Max. Didnt See That be for Monterey 

Intel mbp 2017. 

Only thing I do is hibernate or a restart. I thought it might be the restart but anyway it should go down to 80, Right?

abeldemoz, 2021-11-09:
> Hi @abeldemoz,
> 
> Unfortunately, I am currently not able to reproduce this issue. I have tried it on multiple Intel-based MacBooks with various power adapters and charging past the charge limit never occurred. Do you have any recommendations on how I can reproduce this bug?

I'm not sure how to reproduce it by using a power adapter. I also experience this issue on occasion when using the MacBook Pro in clamshell mode when connected to a monitor via USB-C. The monitor charges the MBP via the USB-C cable. I see it go past the specified limit while I'm using it – the computer is not asleep.

MatthiasKerbl, 2021-11-09:
> > Hi @abeldemoz,
> > Unfortunately, I am currently not able to reproduce this issue. I have tried it on multiple Intel-based MacBooks with various power adapters and charging past the charge limit never occurred. Do you have any recommendations on how I can reproduce this bug?
> 
> I'm not sure how to reproduce it by using a power adapter. I also experience this issue on occasion when using the MacBook Pro in clamshell mode when connected to a monitor via USB-C. The monitor charges the MBP via the USB-C cable. I see it go past the specified limit while I'm using it – the computer is not asleep. I'm not sure if this will help, but maybe the bug could be a result of different wattages from different power supplies. I'm using AlDente-Pro.

Do you have the feature Heat Protection activated? We found a bug in the current version which results in charging above the charge limit for some users if Heat Protection is activated. If you had Heat Protection enabled, you could try it again with Heat Protection disabled.

tosbsas, 2021-11-09:
Will try

tosbsas, 2021-11-09:
I did turn off heat protection, mbp charged to 80% (limit being set by Ardente) and after a while its 81 :-((

abeldemoz, 2021-11-09:
> Do you have the feature Heat Protection activated? We found a bug in the current version which results in charging above the charge limit for some users if Heat Protection is activated. If you had [...]

Yes I have heat protection enabled all of the time. Do you have a timeline for fixing this bug?

MatthiasKerbl, 2021-11-10:
> Yes, I have Heat Protection enabled all of the time. Do you have a timeline for fixing this bug?

We are currently testing a beta version of AlDente which already has the Heat Protection fix implemented. If everything is going well, we will release it soon.

MatthiasKerbl, 2021-11-15:
> Yes, I have Heat Protection enabled all of the time. Do you have a timeline for fixing this bug?

Hi @abeldemoz,

We have just released the new version of AlDente (1.10). https://github.com/davidwernhart/AlDente/releases/
It has a fix for the Heat Protection bug and please try this version, it might have fixed your issue regarding charging past the Charge Limit with high power chargers.

Cheers,
Matthias

jameszhan9592, 2021-11-16:
Hi @MatthiasKerbl ,

I'm on 1.10, but the battery still always gets charged 1 or 2 percent past the specified percentage. Sometimes the charging would even pause at 2% **below** the specified percentage.

Any idea what caused this and how to resolve it?

Best,
James

MatthiasKerbl, 2021-11-17:
> Hi @MatthiasKerbl ,
> 
> I'm on 1.10, but the battery still always gets charged 1 or 2 percent past the specified percentage. Sometimes the charging would even pause at 2% **below** the specified percentage.
> 
> Any idea what caused this and how to resolve it?
> 
> Using an M1 Max MBP, macOS 12.0.1.
> 
> Best, James

Hi @jameszhan9592,

I would say, 1,2,3% is in the margin of error since the battery percentage is not the most accurate number in macOS.

Cheers,
Matthias

tosbsas, 2021-11-17:
It definitivly is. But it was more exacto in earlier versiona 

______________________________
Ruben Gutknecht
Lima, Perú  🇵🇪 
Cel: +51-99349-1515
Voice/Fax: 49 (0) 721-151-379204
http://telegram.me/tosbsas
paypal.me/RubenGutknecht
***@***.***

reednaaxeelaa, 2021-11-23:
Mine continues charging past the specified threshold.
[image]
Coconut battery shows the MacBook currently charging at ~28W, which is the charging rate at when it is close to full.

Sometimes it turns on and sometimes it turns off. I am not sure what triggers it. Over time, it gradually charges fully up. This is while the computer is awake.
I have tried reinstalling the helper application.

Macbook 2021 14 M1 pro.

MatthiasKerbl, 2021-11-23:
> Mine continues charging past the specified threshold. [image] Coconut battery shows the MacBook currently charging at ~28W, which is the charging rate at when it is close to full.
>
> Sometimes it [...]

Hi @reednaaxeelaa,

Do you have "Optimized battery charging" in the macOS preferences disabled?

Cheers,
Matthias

reednaaxeelaa, 2021-11-23:
>
> Hi @reednaaxeelaa,
> 
> Do you have "Optimized battery charging" in the macOS preferences disabled?
> 
> Cheers, Matthias

I do not.

MatthiasKerbl, 2021-11-23:
> > Hi @reednaaxeelaa,
> > Do you have "Optimized battery charging" in the macOS preferences disabled?
> > Cheers, Matthias
> 
> I do not.

In order for AlDente to work, you have to disable this setting.

reednaaxeelaa, 2021-11-23:
Ohh, I am sorry. I misread what you wrote. I do not have the setting enabled.

MatthiasKerbl, 2021-12-11:
> Ohh, I am sorry. I misread what you wrote. I do not have the setting enabled.

Did you try a full reinstall of AlDente?

MatthiasKerbl, 2022-01-18:
> I have a 13-inch 2020 MacBook Pro. When I use the 61W power adapter, everything is fine. When I use a 96W power adapter, it charges past the limit I specified. I set the limit to 80% when connected to the 96W adapter and it charged to 91%.

Hi @abeldemoz,

Do you still experience this issue with our newest version?

closed 2022-02-25 by MatthiasKerbl