AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #440: Battery getting hot

hashsi99, 2021-11-05:
Used AlDente on an intel Mac and never had a problem (classic). 

Just downloaded and started using on a m1 MacBook Pro. Set limit to 77% and it works as it should. However my battery temp (and laptop) are getting warmer then Ever before. I uninstalled and temps are back to normal.  

Is this normal??  I checked the CPU usage with aldente installed and nothing stood out. I was using 1.08 when the overheating started. 

Thanks

MatthiasKerbl, 2021-11-06:
Hi @hashsi99,

AlDente should definitely not cause heat issues with your MacBook because it uses less than 1% of CPU resources.
You could try installing our latest version 1.09, maybe it will fix your issue.

Cheers,
Matthias

hashsi99, 2021-11-06:
I’ll give it a try. Thanks for the quick response. 

The pro does have some battery protection though correct?

MatthiasKerbl, 2021-11-06:
> The pro does have some battery protection though correct?

What exactly do you mean?

hashsi99, 2021-11-06:
https://apphousekitchen.com/feature-explanation-heat-protection/

MatthiasKerbl, 2021-11-07:
> https://apphousekitchen.com/feature-explanation-heat-protection/

Yes, AlDente Pro has the Heat Protection feature which will pause charging when the temperature of the battery is too hot.

MatthiasKerbl, 2021-11-15:
Hi @hashsi99,

Do you still encounter any heat related issues with AlDente?

Otherwise, I would close this issue.

Cheers,
Matthias

hashsi99, 2021-11-15:
you can close it out, I've yet to purchase the pro yet.  But I will soon.  Thanks for the help!

closed 2021-11-15 by MatthiasKerbl