AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #441: Mac M1 sleeps while discharging and small UI Error

gbarcenasjr, 2021-11-06:
I've been using AlDente for months now and didn't have an issue like this.

My 2020 M1 MacBook Pro sleeps while discharging. Normally, that doesn't sound bad but it also stops the discharging leaving me at the same percentage right before the mac slept.

Also, the label above the charge limit percentage now shows "Charge Li..." instead of "Charge Limit". I'm assuming the label's width has been shortened and cutting off the rest of the label text.

[image]

simotau, 2021-11-06:
> My 2020 M1 MacBook Pro sleeps while discharging. Normally, that doesn't sound bad but it also stops the discharging leaving me at the same percentage right before the mac slept.

Hi! Sorry to bother you, but why what you wrote here is supposed to be an issue? I mean, I have installed 1.09 too and I'm expecting that when I put my Mac M1 on sleep it holds the charge until I use it again. 

I probably do not understand well, 'cause I'm not a native english speaker, but I would be more upset if, while sleeping, the Mac discharges instead of holding the charge. 

Can you explain this issue in more details? 'cause I had the 1.08 and everything worked great.

Thank you very much!

MatthiasKerbl, 2021-11-06:
> I've been using AlDente Pro for months now and didn't have an issue like this.
> 
> (Using current version 1.09)
> 
> My 2020 M1 MacBook Pro sleeps while discharging. Normally, that doesn't sound bad but it also stops the discharging leaving me at the same percentage right before the mac slept.
> 
> Also, the label above the charge limit percentage now shows "Charge Li..." instead of "Charge Limit". I'm assuming the label's width has been shortened and cutting off the rest of the label text.
> 
> [image]

Hi @gbarcenasjr,

If you don't use your MacBook it will automatically go into sleep even if Discharge is enabled in AlDente. We might include the option of disabling sleep until the Charge Limit is reached in a future update.

Thank you for pointing out the Charge Limit label issue. We will fix it with the next update again.

Cheers,
Matthias

closed 2021-11-06 by MatthiasKerbl

gbarcenasjr, 2021-11-08:
Hello @simotau and @MatthiasKerbl ,

Sorry if my last post was hard to understand. I'll try to explain a bit more. 

I'm not used to the Mac sleeping while discharging. Sleeping while using AlDente to discharge used to be disabled before I updated AlDente on my Mac. When I saw that my Mac when to sleep the first time, I thought it was an error, but I see it's actually a feature. Sorry about the confusion.

My issue was that once the Mac goes to sleep, AlDente pauses the discharge rather than letting the Mac discharge to the specified limit. And when I wake up the Mac, AlDente doesn't return to Discharge. I have to press Discharge again and every time I wake up my computer until I reach my charge limit.

I don't know if that helps, but feel free to let me know if you're still confused. Thank you for taking the time to read my post!

MatthiasKerbl, 2021-11-09:
Hi @gbarcenasjr,

Thank you for your more detailed explanation. Right now, Discharge is intended for the following scenario: You are plugging in your MacBook with a 90% state of charge. Your charge limit is set to 80%, therefore, you want to discharge it to 80%. You activate Discharge, while you continue using your MacBook (otherwise, it would barely lose any battery). When 80% is reached, AlDente automatically stops charging.

We are currently working on the whole "stop sleeping" situation, so stay tuned.

Cheers,
Matthias

gbarcenasjr, 2021-11-16:
Hey @MatthiasKerbl ,

I just saw the new update and wanted to thank you and the team for the "Stop Sleeping" feature! Now my mac is running as it used to and now I don't have to be at my desk to have my mac discharge the battery.

Also, the "Charge Limit:" Label looks great again!

MatthiasKerbl, 2021-11-16:
> Hey @MatthiasKerbl ,
> 
> I just saw the new update and wanted to thank you and the team for the "Stop Sleeping" feature! Now my mac is running as it used to and now I don't have to be at my desk to have my mac discharge the battery.
> 
> Also, the "Charge Limit:" Label looks great again!

Hi @gbarcenasjr,

I am glad you like the new feature. It was something we were looking forward to implementing.

Cheers,
Matthias