AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #444: Would it be possible to have max-charge % respond to an external app?

DamonHD, 2021-11-07:
Hi,

It would be very useful to have AlDente periodically poll for a specified external file (or URL), and if it exists and contains a single integer between (say) 20 and 100 use that as the charge limit.

That would let me auto-adjust adjust my MacBook Air to play better with my off-grid system, and also with the grid (eg using a lower max charge level when the grid is dirty).

See:
    http://www.earth.org.uk/Hey-Siri-Help-Flatten-The-Duck.html
    http://www.earth.org.uk/_gridCarbonIntensityGB.html

I am in the UK.

Rgds

Damon

MatthiasKerbl, 2021-11-08:
Hi @DamonHD,

This looks very interesting as well. We will also discuss it and I will come back to you.

Cheers,
Matthias

DamonHD, 2022-08-02:
See https://twitter.com/andysc/status/1554378086459785217

closed 2022-09-01 by MatthiasKerbl