AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #445: Would it be possible to automatically reduce the maximum charge level at local peak time?

DamonHD, 2021-11-07:
Hi,

Something that might help grids around the world would be reducing the max-charge percentage threshold temporarily between (say) 4pm and 7pm (ish) local time.

Maybe this feature could drop the maximum charge percentage from the user-set value by 10% (or even down to 50% or 20%) between those hours.  This could simply reduce/prevent charging during these times, or could even be in 'discharge' mode if need be to try to avoid using power via the adaptor at all during these hours.

Then AlDente users will by default help 'shave the peak' on their local grid, and probably also charge with less 'dirty' electricity.

Rgds

Damon

MatthiasKerbl, 2021-11-08:
Hi @DamonHD,

This sounds like an interesting additional feature for AlDente Pro. It could be called "Green Mode" or "Peak Shaver Mode". We will discuss this as a potential feature in a future update.
Do you have some recourses regarding the worldwide peak grid times?

Cheers,
Matthias

DamonHD, 2021-11-08:
Very broadly 4pm to 9pm (later afternoon to early evening) will be peak time in many markets much of the year.  For the GB grid 4pm to 7pm local time is also when a lot of the distribution companies slap on a premium to try to deter usage.

Looking at #444 cleverer and more dynamic things can be done, but this seems a simple starting point.

I can go and talk to some contacts in the GB grid and see if they agree with my assertions...  B^>

Rgds

Damon

DamonHD, 2021-11-08:
One comment so far from a grid person:

> On the whole it’s a great idea to demand shift away from the peak.

So, maybe start with something simple (4pm-9pm) and then add more bells and whistles if people like/use it.

I would suggest making at least the simple version available in the Free app, to make it free to help save the planet a bit!

Rgds

Damon

DamonHD, 2021-11-08:
FYI:

https://energymag.net/daily-energy-demand-curve/

Note morning peaks in some many places too.

Rgds

Damon

PearApple151, 2021-11-19:
love the idea, I'm just curious. does my laptop charging make that much of a difference? I understand something like a washing machine or an EV.

DamonHD, 2021-11-19:
Millions of laptops each with tens of watt.hours (Wh) of storage (~200M ship per year globally) makes for at least of 10s of megawatt.hours (MWh) of potential deferred/shifted demand.  Large grid-connected battery such as Australia's Hornsdale Power Reserve is ~200MWh, so on a global scale it is visible.

Another way of viewing it from the UK (GB grid) perspective: more that half the population has a laptop (https://www.statista.com/statistics/274137/laptop-penetration-in-the-united-kingdom-uk-since-2009/ thus maybe 30M laptops), and typical charging power may be 20W, so deferrable charging power ~600MW.  Compared to peak demand on the GB grid yesterday of 40GW, or total contracted deferrable demand response of ~2GW IIRC, that is again significant.  It won't prop the grid up on its own, but it essentially costs nothing (CAPEX) and doesn't hurt (may even prolong battery life), and is potentially a sizeable fraction of existing solutions.

Rgds

Damon

DamonHD, 2021-11-19:
> love the idea, I'm just curious. does my laptop charging make that much of a difference? I understand something like a washing machine or an EV.

You should also avoid running your washing machine or charging your EV in peak hours.  Deferring charging your laptop should be additive not a replacement.

Rgds

Damon

doyled74, 2022-03-03:
Registered for GitHub just to indicate my support and comment for this feature request.

I've just downloaded AlDente as I was looking for a 'Coast' or similar type feature that would enable the laptop to run off battery during the highest CO2 intensity periods of the UK grid, and to recharge when it is lowest CO2 intensity. The use of electricity during low intensity/low demand periods is already incentivised in the UK and other countries through Time Of Use electricity tariffs.

UK CO2 Grid Intensity is published publically and already incorporated into smartphone applications : https://carbonintensity.org.uk

[image]
[image]

In the UK Time Of Use tariffs mean that an overnight electricity rate (typically 00:30-04:30, but varies) is around 1/3rd the price of daytime rate.

@DamonHD has already produced some excellent stats and justification. On a modern laptop with a large 100 Wh battery say the financial saving from overnight charging is small compared to running large domestic appliances, but it's the impact on the grid that is the key benefit and having the ability to schedule my laptop in the same way as I can schedule a dishwasher or washing machine. 

If I could have a schedule that switches to 'Discharge' mode at say 15:00, I run on battery until I finish working at 17:00 and then the laptop charges again either overnight or from 07:00 the next morning, that is certainly of interest. Customising within the app should be possible so users can set the behaviour based on their local grid.

doyled74, 2022-03-03:
> Hi @DamonHD,
> 
> This sounds like an interesting additional feature for AlDente Pro. It could be called "Green Mode" or "Peak Shaver Mode". We will discuss this as a potential feature in a future update. Do you have some recourses regarding the worldwide peak grid times?
> 
> Cheers, Matthias

Here is an excellent example of the difference in UK CO2 grid intensity during a day (yesterday) and showing same pattern shown over a month.

https://electricityinfo.org/carbon-intensity-archive/
[image]
[image]

DamonHD, 2022-06-29:
For my new M1 MacBook Air, turning on low power mode at the same time as inhibiting charging would seem to likely save another ~0.7W/15%.

There are maybe 2 million MacBook users on the GB grid.

http://www.earth.org.uk/Hey-Siri-Help-Flatten-The-Duck.html#Measured

Rgds

Damon

DamonHD, 2022-08-02:
See also this from the CTO of IBM UK:

https://twitter.com/andysc/status/1552232493708967941

MatthiasKerbl, 2022-08-30:
Hi @DamonHD,

Not (yet) the feature you were asking but with the newest version (AlDente Pro 1.18) we have added support for Apple Shortcuts to AlDente. This enables you to trigger AlDente features like the Charge Limiter with other apps as well and would theoretically give you the option to connect it to a different app that has the local peak times.
You can read more about the Shortcuts support here: https://apphousekitchen.com/feature-explanation-shortcuts/
Also, this is only a beta release, bugs will probably still occur.

Cheers,
Matthias

DamonHD, 2022-08-30:
Ah, thank you.  Hadn't realised.

I'll give myself some time to play with it next week!

Rgds

Damon

DamonHD, 2023-03-07:
Any progress on this, built-in, along with the lowpowermode?

Rgds

Damon

PS. No, I haven't had the time to tool up to use the Shortcuts yet.  %-P

MatthiasKerbl, 2023-03-12:
> Any progress on this, built-in, along with the lowpowermode?
> 
> Rgds
> 
> Damon
> 
> PS. No, I haven't had the time to tool up to use the Shortcuts yet. %-P

This potential future feature is in our backlog but not yet a priority since other features and improvements are more important at the moment. However, the new Schedule feature can already help and makes it possible to implement a first version. Furthermore, connecting AlDente with an external "CO2 intensity app" through the Shortcuts app is most likely also possible.
Adding support for the low-power mode is probably easier and will therefore happen sooner.

Hope that's understandable.

Cheers,
Matthias

DamonHD, 2023-03-12:
I'm happy with a fixed schedule for now, and I've already put one in place using 1.21.  My next request would indeed be lowpowermode support so I can remove my cron kludge!

Thank you!

Rgds

Damon

http://www.earth.org.uk/Hey-Siri-Help-Flatten-The-Duck.html#2023-03-11

DamonHD, 2023-03-15:
FYI: https://www.washingtonpost.com/climate-environment/2023/03/14/clean-energy-charging-iphone-apple/

DamonHD, 2024-09-18:
Your low power mode support is good, thank you.

Please do add control of it to the scheduled tasks as an option for each task so that I can use it instead of my crontab kludge as above.

The new feature might be a three-way option:
  * (default) leave low power mode alone
  * force low power mode on
  * force low power mode off

Thanks again!

Damon

DamonHD, 2025-06-29:
Thank you for 1.34, which has me a little closer to where I want to be!

https://www.earth.org.uk/Hey-Siri-Help-Flatten-The-Duck.html#2025-06-28

A low-priority suggestion for you:

Looking at the 'Weekdays' section, to provide dynamic response to grid state, which is going to become more important, it might be useful to flip between Charge and Discharge (without changing the threshold) based on grid intensity.  In particular when the grid intensity is high, Discharge, else Charge.  That could play nicely othogonally and race-free against other activities such as changing the charge threshold.

DamonHD, 2026-10-11:
FWIW in this paper I discuss how important demand flexibility will be, and while AlDente isn't going to make or break the grid, it absolutely could help as described in the "Hey Siri" piece too:

https://www.academia.edu/2998-3665/3/3/10.20935/AcadEnergy8514