AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #446: 2021 M1 Pro MacBook Pro, sometime charge to 100% after shut down (limiter on to 55%)

ShapeSh1ft3r, 2021-11-08:
My Mac powered from an LG monitor via USB C. (I don't use HDMI port). The monitor always connected to the MacBook right thunderbolt 4 port.

I've set the charge limit to 55%. While I work on the MacBook everything works fine, charging stays at 55%. In sleep mode ok too. Charge limiter works like a charm.

When I shut down the machine, for example yesterday night with 51%, on the next day morning I've got a battery with 100% charged. Sometimes the limiter works, and the battery level stays where I left.

I did tried to reinstall AIDente, helper...

So, for my case, the charge limiter when the machine turned off, sometimes works, sometimes not. 
Now I discharging my battery to 55 percent with AiDente's discharge function and I will not turn off my machine until this problem exists. I will use only the sleep mode.

More detail:
- Lid not closed
- Sailing Mode -> enabled
- Stop charging when sleeping -> disabled
- Stop charging when app closed -> enabled
- AIdente Pro 1.09
- macOS Monterey 12.0.1

Sorry for my english :)
Thy!

MatthiasKerbl, 2021-11-08:
Hi @ShapeSh1ft3r,

Unfortunately, AlDente's ability to control the charging behavior of your MacBook while it is in sleep or shut down is highly limited.
Please read the following articles on our blog for more details:

- https://apphousekitchen.com/feature-explanation-stop-charging-when-sleeping/
- https://apphousekitchen.com/feature-explanation-stop-charging-when-powered-off-app-closed/

Hope that clarifies some things!

Cheers,
Matthias

closed 2021-11-08 by MatthiasKerbl

ShapeSh1ft3r, 2021-11-08:
Ohh.
This one explains everything to me.
https://apphousekitchen.com/feature-explanation-stop-charging-when-powered-off-app-closed/

I had Intel based MacBook previously but now...  thx.

MatthiasKerbl, 2021-11-08:
> Ohh. This one explains everything to me. https://apphousekitchen.com/feature-explanation-stop-charging-when-powered-off-app-closed/
> 
> I had Intel based MacBook previously but now... thx.

My suggestion for you would be to either not shut down your MacBook and enable "Stop charging when sleeping" or to unplug your MacBook during the night and shut it down.