AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #450: [Thread] Translating AlDente

MatthiasKerbl, 2021-11-09:
Hi Everyone,

Following a recipe in a foreign language can't be a good idea. That's why we started translating AlDente into other languages too.

AlDente is currently available in:
- English
- German

Languages currently under construction:

- 

Do you want to help us translate AlDente into more languages to help users all over the world? 
Follow these steps:

- Comment on which language you want to translate AlDente into.
- Download the language file of AlDente here (https://gofile.io/d/uc1Biw)
- Open the file with a text editor of your choice and translate AlDente in your language.
- Upload the translated and renamed file here (https://gofile.io/welcome) or use a filesharing platform of your choice.
- Share the download link in the comments below.

Thank you so much and greetings from the kitchen,

David and Matthias

HeySora, 2021-11-10:
Hey,

Here is a fully translated french file. :)
I've done my best to keep the length of short messages (like button labels) as similar as possible while keeping the original wording. However do not hesitate to ping me if a specific text is too big!

https://priv.heysora.net/AlDente%20French.strings

MatthiasKerbl, 2021-11-10:
> Hey,
> 
> Here is a fully translated french file. :) I've done my best to keep the length of short messages (like button labels) as similar as possible while keeping the original wording. However do not hesitate to ping me if a specific text is too long!
> 
> https://priv.heysora.net/AlDente%20French.strings

Hi @HeySora,

Thank you so much, we will include it with the next update. :)

Cheers,
Matthias

simotau, 2021-11-10:
Currently working on Italian version. One question:

"lblStopChargingWhenClosed" = "Stop charging\nwhen app closed";
"lblStopChargingWhenPoweredOff" = "Stop charging\nwhen powered off";

Are these two the same? 'cause as far as I know there's no option to limit the charge while the MacBook is powered off.

MatthiasKerbl, 2021-11-10:
> Currently working on Italian version. One question:
>
> "lblStopChargingWhenClosed" = "Stop charging\nwhen app closed"; "lblStopChargingWhenPoweredOff" = "Stop charging\nwhen powered off";
>
> Are these [...]

Hi @simotau,

Thank you so much for helping us.

This feature is different depending on which MacBook model you use. In short: Apple silicon MacBooks don’t have the ability to pause charging when they are powered off. Therefore, the feature is called "Stop charging when app closed". On Thunderbolt 3 Intel MacBooks, on the other hand, charging can be paused while they are turned off. Therefore, the feature is called "Stop charging when powered off". You can read more about those features and what they do in detail in our blog post here: https://apphousekitchen.com/feature-explanation-stop-charging-when-powered-off-app-closed/

Cheers,
Matthias

simotau, 2021-11-10:
Hey!

I tried my best to do a decent job: https://gofile.io/d/aUtv55

I left some word like "Top Up" or "Helper" the same. The first one 'cause I think it's a cool way to name a feature, the second one in order to make easier to people to find the application helper into the Library folder if they want to.

I hope my Italian fellows won't be too judgmental to me 😅

Bye,
Simone.

GZaccaroni, 2021-11-11:
> Hey!
>
> I tried my best to do a decent job: https://gofile.io/d/aUtv55
>
> I left some word like "Top Up" or "Helper" the same. The first one 'cause I think it's a cool way to name a feature, the second [...]

It seems good to me!

MatthiasKerbl, 2021-11-11:
> Hey!
>
> I tried my best to do a decent job: https://gofile.io/d/aUtv55
>
> I left some word like "Top Up" or "Helper" the same. The first one 'cause I think it's a cool way to name a feature, the second [...]

Thanks a lot, @simotau, we will include it with the next update.

Cheers,
Matthias

tosbsas, 2021-11-11:
Spanisch still needed?

MatthiasKerbl, 2021-11-12:
> Spanisch still needed?

Yes, Spanish is still needed.

tosbsas, 2021-11-12:
ok, going to do spanisch

tosbsas, 2021-11-12:
here you go

https://we.tl/t-A7hN6PoKpb

Spanisch

MatthiasKerbl, 2021-11-12:
> here you go
> 
> https://we.tl/t-A7hN6PoKpb
> 
> Spanisch

Thanks a lot, @tosbsas.

huang-1331, 2021-11-16:
Here is Korean translations!

https://gofile.io/d/DcxgkZ

MatthiasKerbl, 2021-11-16:
> Here is Korean translations!
> 
> https://gofile.io/d/DcxgkZ

Thank you so much @huang-1331!

withduan, 2021-11-16:
This is Chinese Simplified translations.
All content has been translated, except for the software copyright information.
https://gofile.io/d/cdbQzO

MatthiasKerbl, 2021-11-16:
> This is Chinese Simplified translations. All content has been translated, except for the software copyright information. https://gofile.io/d/cdbQzO

Thank you so much @withduan.

sluttyscott, 2021-11-16:
Hi Matthias,
Although Hwang did a great job on his/her translation, but I found some parts are have to be improved.
So I send you a version of mine anyways. 
https://gofile.io/d/YIxRpo
I brought specific words from Korean version of macos. And applied it the translation into real app and made some adjustments on line breaking.(But I was not able to test it in all scenarios like installing) In this way the reading will be much natural.
One thing I worry about is the sailing mode part, it is not that natural and idk if I really understood this feature. If there are some mistakes, I hope someone correct them later.
Thanks a lot for cooking an awesome app!

sluttyscott, 2021-11-16:
Hi Matthias,
Although Hwang did a great job on his/her Korean translation, but I found some parts are have to be improved.
So I send you a version of mine anyways.
https://gofile.io/d/YIxRpo
I brought specific words from Korean version of macos. Also applied it into real app and made some adjustments on line breaking and word choices. (But I was not able to test it in all scenarios like installing) In this way the reading will be much natural.
One thing I worry about is the sailing mode part, it is not that natural and idk if I really understood this feature. If there are some mistakes, I hope someone correct them later.
Thanks a lot for cooking an awesome app!

sluttyscott, 2021-11-16:
Hi Matthias,
Although Hwang did a great job on his/her Korean translation, but I found some parts are have to be improved.
So I send you a version of mine anyways.
https://gofile.io/d/oXU7rx
I brought specific words from Korean version of macos. Also applied it into real app and made some adjustments on line breaking and word choices. (But I was not able to test it in all scenarios like installing) In this way the reading will be much natural.
One thing I worry about is the sailing mode part, it is not that natural and idk if I really know about this feature. If there are some mistakes, I hope someone correct them later.
Thanks a lot for cooking an awesome app!

MatthiasKerbl, 2021-11-16:
> Hi Matthias, Although Hwang did a great job on his/her Korean translation, but I found some parts are have to be improved. So I send you a version of mine anyways. https://gofile.io/d/oXU7rx Specific words are from Korean version of macos. Also applied it into real app and made some adjustments on line breaking and word choices. (But I was not able to test it in all scenarios like installing) In this way the reading will be much natural. One thing I worry about is the sailing mode part, it is not that natural and idk if I really know about this feature. If there are some mistakes, I hope someone correct them later. Thanks a lot for cooking an awesome app!

Hi @scott-cho,

Thanks a lot!

brucruz, 2021-11-16:
Great job! Do you need a Portuguese-BR version

MatthiasKerbl, 2021-11-16:
> Great job! Do you need a Portuguese-BR version?

Hi @brucruz,

Yes, we still need a Portuguese version.

ammarfatani, 2021-11-16:
Hi,
I really like the app and would like to contribute to translating it into Arabic if you like me too. 
Arabic letters are encoded differently than other languages and are read from right to left. Should I just replace the text in the string file directly or do I need to do something special with it?

MatthiasKerbl, 2021-11-17:
> Hi, I really like the app and would like to contribute to translating it into Arabic if you like me too. Arabic letters are encoded differently than other languages and are read from right to left. [...]

Hi @ammarfatani,

Thank you for helping us. For now, you can just replace the text in the file.

dionvl, 2021-11-18:
Hi, I have translated the file to Dutch (nl-NL / nl-BE). I've tired to stick as close to macOS native wording as possible. Everything should be OK, but a second pass by someone else is always appreciated. :) 

https://gofile.io/d/hxPiNz

Also, I have noticed there are no strings for the update or license screens (?)

MatthiasKerbl, 2021-11-18:
> Hi, I have translated the file to Dutch (nl-NL / nl-BE). I've tired to stick as close to macOS native wording as possible. Everything should be OK, but a second pass by someone else is always [...]

Hi @dionvl,

Thank you so much. :)

MatthiasKerbl, 2021-11-18:
> Hi, I have translated the file to Arabic (AR) as promised. Can you please check the file in the below link and tell me if any further update is needed.
> 
> https://gofile.io/d/y6QqV3
> 
> Regards,

Thanks a lot @ammarfatani!

vingchung, 2021-11-20:
[https://gofile.io/d/R4cohO](url)

Traditional Chinese is here! 
Tell me if any problem or further update is needed. 
Thank you for your excellent app !

MatthiasKerbl, 2021-11-20:
> [https://gofile.io/d/R4cohO](url)
> 
> Traditional Chinese is here! Tell me if any problem or further update is needed. Thank you for your excellent app !

Thanks you so much @vingchung!

KonradDIKU, 2021-12-02:
"Download the language file of AlDente here (https://gofile.io/d/tJ6WDZ)" - link expiered

MatthiasKerbl, 2021-12-02:
> "Download the language file of AlDente here (https://gofile.io/d/tJ6WDZ)" - link expiered

Hi @KonradDIKU,

Thanks for the reminder, I just updated the link.

Cheers,
Matthias

spacemachine3924, 2021-12-06:
I really enjoy the app as I’m spending all my time with my MacBook at home, so here is a translation in Greek!

https://gofile.io/d/VF5PZi

Any chance those of us contributing to a translation could get a pro license key as a courtesy? No worries if not, keep up the good work!

KonradDIKU, 2021-12-09:
Polish is here! https://gofile.io/d/liQ4SS

All should be fine, at least understandable to every native speaker! ;)

Patryk122PL, 2021-12-11:
Hi, native Polish speaker here!

I've fixed all little and contextual mistakes in polish translation by @KonradDIKU (btw. great job Konrad! I assume polish isn't your native language and yet I'm impressed by overall pretty accurate translation - thank you for making it).

Download: [https://gofile.io/d/Mc23VP](url)

**Side note:**
In english example file there are two mistakes in what supposed to be compared to what description says. These two are: _"hlpShowPercentage"_ and _"hlpShowBatteryIcon"_. Their description is just the same as for _"hlpShowOnlyAlDenteIcon"_. 

I've provided correct translation in my file, but you may want to update this in OP since it might be overlooked by other people.

Cheers,
Patryk

MatthiasKerbl, 2021-12-11:
> I really enjoy the app as I’m spending all my time with my MacBook at home, so here is a translation in Greek!
> 
> https://gofile.io/d/VF5PZi
> 
> Any chance those of us contributing to a translation could get a pro license key as a courtesy? No worries if not, keep up the good work!

Hi @spacemachine3924,

Thank you so much for the Greek translation.
Since so many people offered to translate AlDente even before we introduced the option of translations, we did not think about offering AlDente Pro for free. Hope that's understandable.

Cheers,
Matthias

MatthiasKerbl, 2021-12-11:
> Polish is here! https://gofile.io/d/liQ4SS
> 
> All should be fine, at least understandable by an every native speaker! ;)

Hi @KonradDIKU,

Thanks a lot for the Polish translation.

Cheers,
Matthias

MatthiasKerbl, 2021-12-11:
> Hi, native Polish speaker here!
> 
> I've fixed all little and contextual mistakes in polish translation by @KonradDIKU (btw. great job Konrad! I assume polish isn't your native language and yet I'm impressed by overall pretty accurate translation - thank you for making it).
> 
> Download: https://gofile.io/d/Mc23VP
> 
> **Side note:** In english example file there are two mistakes in what supposed to be compared to what description says. These two are: _"hlpShowPercentage"_ and _"hlpShowBatteryIcon"_. Their description is just the same as for _"hlpShowOnlyAlDenteIcon"_.
> 
> I've provided correct translation in my file, but you may want to update this in OP since it might be overlooked by other people.
> 
> Cheers, Patryk

Hi @Patryk122PL,

Thanks for the improvements.
And you are right, there is something off with the descriptions you mentioned. We will look into this.

Cheers,
Matthias

vingchung, 2022-03-15:
[](https://gofile.io/d/qubzqS)
Traditional Chinese Translation updated, just revised some words for better UI, THX.

vingchung, 2022-03-15:
https://gofile.io/d/OI477L
Traditional Chinese Translation updated, just revised some words for better UI, THX.

RodrigoGCoelhoo, 2022-04-30:
Hey guys! I would love to translate AlDente to Portuguese (Brazil), but I'm having some trouble to open the links. It displays a message saying that the file does not exist.

MatthiasKerbl, 2022-05-11:
> Hi, In AlDente version 1.16 one line got changed under general -> updates: "Check Automatically" should be in Polish: "Sprawdź automatycznie".
> 
> Side note: the link to English language file from OP is no longer available to download.

Hi @Patryk122PL,

Thanks for telling us, we will fix it with the next update.

Cheers,
Matthias

claudewowo, 2022-10-17:
I have fixed some mistake in Chinese(simplified)
https://gofile.io/d/j3CGZY

MatthiasKerbl, 2022-10-17:
> I have fixed some mistake in Chinese(simplified) https://gofile.io/d/j3CGZY

Hi @claudewowo,

Thank you for your help, we will implement it with the next update.

Cheers,
Matthias

claudewowo, 2022-10-28:
I found some words in English, so I translated to Chinese(simplified), please update, thanks.
https://gofile.io/d/L5NkNI

And may I ask for a pro licence?

MatthiasKerbl, 2022-10-28:
> I found some words in English, so I translated to Chinese(simplified), please update, thanks. https://gofile.io/d/L5NkNI
> 
> And may I ask for a pro licence?

Hi @claudewowo,

Thank you so much for improving the Chinese(simplified) translation.
Since so many people offered to translate AlDente even before we introduced the option of translations, we did not think about offering AlDente Pro for free. Hope that's understandable.

Cheers,
Matthias

RodrigoGCoelhoo, 2022-11-10:
Hello,
Here is the translation for Portuguese (Brazil).

https://gofile.io/d/teRKT2

xhable1337, 2022-12-08:
Hi,
I am Russian native speaker, and I would like to fix translation, bc in some cases it's very complicated and wrong, but link in OP is restricted. Am I supposed to request an access through the Google's form?

Patryk122PL, 2023-01-02:
Hi,
I  would like to suggest an improvement into translating process.
Could you upload to GitHub already translated files so we can do pull requests ourselves?

Basically I submitted Polish translation a year ago and since then quite a lot has changed in AlDante. I wanted to update my translation but lost the file so now we have some aspects of the app quite well translated mixed with small errors and a few mistranslated lines.

Cheers,
Patryk

saxAllan, 2023-02-16:
Hi @MatthiasKerbl 
I am trying to create a Japanese translation, but I don't seem to have access to the English files.
Could you please fix this?

MatthiasKerbl, 2023-03-03:
> Hi @MatthiasKerbl I am trying to create a Japanese translation, but I don't seem to have access to the English files. Could you please fix this?

Hi @saxAllan,

Thank you for helping us to translate AlDente. You can now download the English language file in the original post on top.

Cheers,
Matthias

MatthiasKerbl, 2023-03-03:
> Hi, I would like to suggest an improvement into translating process. Could you upload to GitHub already translated files so we can do pull requests ourselves?
>
> Basically I submitted Polish [...]

Hi @Patryk122PL,

Thank you for helping us with the translations.
You can already access the newest language files on your MacBook when you have AlDente installed.
Just follow these steps: "Open the language file by right clicking on the AlDente app file in the Applications folder and choosing "Show Package Contents". Navigate to Contents -> Resources -> and choose the language folder of your choice. Copy the Localizable.strings file and make any improvements you can find. Afterward, upload the new file in the comments below."

Cheers,
Matthias

MatthiasKerbl, 2023-03-03:
> Hi, I am Russian native speaker, and I would like to fix translation, bc in some cases it's very complicated and wrong, but link in OP is restricted. Am I supposed to request an access through the Google's form?

Hi @xhable1337,

Thank you for helping us improving the translations.
You can always access the newest language file on your MacBook when you have AlDente installed.
Just follow these steps: "Open the language file by right clicking on the AlDente app file in the Applications folder and choosing "Show Package Contents". Navigate to Contents -> Resources -> and choose the language folder of your choice. Copy the Localizable.strings file and make any improvements you can find. Afterward, upload the new file in the comments below."

Cheers,
Matthias

MatthiasKerbl, 2023-03-12:
> Hi @MatthiasKerbl
> 
> At first, I made a Japanese translation of the file. [Japanese Default.txt](https://github.com/davidwernhart/AlDente-Charge-Limiter/files/10951253/Japanese.Default.txt)
> 
> I am sorry, but I have never used pro's unique features, so I'm a little worried about the translations regarding them. If I ever use Pro in the future, I hope to improve it then.

Hi @saxAllan,

Thanks a lot for the translation, we will add it in one of the upcoming updates.

Cheers,
Matthias

Patryk122PL, 2023-03-15:
Ok.. finally found some time to update my Polish translation.
Hope it helps!
[pl.lproj.zip](https://github.com/davidwernhart/AlDente-Charge-Limiter/files/10978603/pl.lproj.zip)

keman5, 2023-03-17:
I have fixed some mistake in Chinese(simplified).
[zh-Hans.Default.strings.zip](https://github.com/davidwernhart/AlDente-Charge-Limiter/files/10999050/zh-Hans.Default.strings.zip)

issac4892, 2023-03-27:
[Localizable.strings.zip](https://github.com/davidwernhart/AlDente-Charge-Limiter/files/11075647/Localizable.strings.zip)
I fixed some non-natural strings

furkanipek, 2023-08-28:
[tr-Default.strings.zip](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/files/12451141/tr-Default.strings.zip)
Turkish translation has been added.

Terabyte1385, 2023-08-28:
[Romanian.zip](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/files/12455799/Romanian.zip)
Hi @MatthiasKerbl,

Here's a full Romanian localization. I have no idea if it's needed but if it is I'm more than glad to help!

Please note that I've attempted to properly match Apple's system translations, but it ain't perfect. I think users may have a difficult time trying to understand some of Apple's stupid translations. What I'm trying to say is that it doesn't 100% match Apple's system translations. Please let me know if you want that, and I'll update accordingly.

BaikKuma, 2023-11-25:
[Korean.zip](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/files/13463229/Korean.zip)

I saw multiple Korean users have been contributed to the language and was mostly satisfied with the translation. But there were some off translation on the scheduling task part of the translation and fixed a few.
Also seems like the contribution was about couple years ago so wish this fix helps more Korean easy to use AlDente as I do!

MilesTEG1, 2023-11-29:
Hello, 
I corrected some small typo in French (a space before `:`), and some translation.
Homogenizing the charge limit translation to "Limite de chargement".
[fr.lproj.zip](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/files/13497991/fr.lproj.zip)

[15 more replies]

closed 2026-10-13 by MatthiasKerbl