AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #451: License checking issue

polarimetric, 2021-11-11:
AlDente is complaining that I have "no connection," saying that it "needs to check if my license is valid once every 10 days." My internet connection is otherwise functional. Because AlDente thinks I don't have an internet connection, my Pro license features have been deactivated. I'm running 1.0.9 and this just started happening tonight. Is there an issue with whatever server is used for the license checking?

polarimetric, 2021-11-11:
An update: I appear to have resolved this issue for now by completely removing AlDente, reinstalling it, and re-entering my license information. Since no one else has raised the same issue, it's possible this was just a bizarre fluke on my end. I did recently get a new laptop, so it's possible something weird happened with the migration.

MatthiasKerbl, 2021-11-12:
> An update: I appear to have resolved this issue for now by completely removing AlDente, reinstalling it, and re-entering my license information. Since no one else has raised the same issue, it's [...]

Hi @polarimetric,

Thank you for updating us on your issue and I am glad you could fix it. We will look into the licensing system if anything comes up again. For now, I will close this issue but feel free to write us again if anything comes up.

Cheers,

Matthias

closed 2021-11-12 by MatthiasKerbl

Proplex, 2021-11-23:
Hi there, I just want to echo that I had the same exact issue.