AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #453: Aldente

police340, 2021-11-12:
Hello,
Just bought paid version of this program and installed it properly. Its in the application folder but double clicking it does not start the program. Deleted and redownloaded and same results. Suggestions? Thank you, Bill

MatthiasKerbl, 2021-11-15:
> Hello, Just bought paid version of this program and installed it properly. Its in the application folder but double clicking it does not start the program. Deleted and redownloaded and same results. Suggestions? Thank you, Bill

Hi @police340,

Thank you for buying AlDente Pro and I am sorry you are having Problems with it.

Do you see AlDente in Activity Monitor after double clicking on AlDente?

Cheers,
Matthias

police340, 2021-11-15:
It is there but nothing happens when I start the app. David said a new
release is coming. Will that be today?

MatthiasKerbl, 2021-11-15:
> It is there but nothing happens when I start the app. David said a new release is coming. Will that be today?
> […](#)

Hi @police340,

The new version of AlDente got released today, you can download it here: https://github.com/davidwernhart/AlDente/releases

If you still encounter any problems, report back to us through the email support or close the email support for this issue and report back here.

Cheers,
Matthias

closed 2021-11-15 by MatthiasKerbl

police340, 2021-11-15:
Hello,

Just downloaded and install. Still does not do anything. It does show up in
activity monitor.

police340, 2021-11-16:
Hello Gentlemen,

Have you any ideas about this not working as I’m about to dispute the
charge.

Thank you,
Bill

MatthiasKerbl, 2021-11-17:
Hi @police340,

Which MacBook model do you have?
Which macOS version do you use?
Is it possible that your menubar is full and therefore, you can not see the AlDente icon?

police340, 2021-11-17:
Hello,

I have a Macbook Pro 2019 model, OSX Monterey 12.0.1 You have already
mentioned the menue bar and I have checked. That is not the issue. Even if
it were full, I should be able to find your program running somewhere.
Maybe we can do a screen share.?

police340, 2021-11-17:
I just noticed the icon up top, i never looked up there. I always thought
you were talking about the bar at the bottom of screen. Looks good, thank
you.

Bill

police340, 2021-12-21:
Hello,

I received this message and I have the pro version registered. What does it
mean?

Active Discharge in Clamshell Mode is only available with the Pro version.

Thank you, Bill