AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #458: Any solution for older macs than the ones supported?

andreschouhy, 2021-11-15:
Hi, I have a macbook pro 13 from 2009, MacOS 10.11 El Capitan. I understand that AlDente doesn't support my computer. But is there some workaround for my case? Maybe some terminal command I can automate?
Thank you

andreschouhy, 2021-11-15:
By the way, I'm only interested in the Charge Limiter feature.

MatthiasKerbl, 2021-12-22:
Hi @andreschouhy,

The current version of AlDente (1.12) is only compatible with macOS 11 Big Sur. This means it is compatible with all MacBooks since 2013.
AlDente Classic 2.2 is Compatible with macOS 10.15 Catalina. This means it is compatible with all MacBooks since 2012.
AlDente Classic 1.2 might be compatible with older versions of macOS. However, I am not sure. You can download it here: https://github.com/davidwernhart/AlDente/releases?page=2

Cheers,
Matthias

closed 2021-12-22 by MatthiasKerbl