AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #461: Feature Requests: Hot Keys

abeldemoz, 2021-11-16:
AlDente-Pro user here. The app is great, but one gripe I have is that I have to manually move my cursor over to the menu bar icon just to change the charging limit. It would be better if you could let users assign their own hot key or shortcut that does this for them. In other words, I want to be able to change the charging limit without taking my hand off of the keyboard.

MatthiasKerbl, 2021-11-17:
> AlDente-Pro user here. The app is great, but one gripe I have is that I have to manually move my cursor over to the menu bar icon just to change the charging limit. It would be better if you could [...]

Hi @abeldemoz,

Good idea, we might be able to do something about that.

Cheers,
Matthias

PearApple151, 2021-11-19:
would be cool to be able to incorporate it into the touch bar somehow. a hotkey activates the menu, and then I can just adjust the slider like I do screen brightness or volume??? Not sure if that exists in the realm of possibility. haha

skynet01, 2022-01-27:
Another +1 for the hotkeys. Would be also nice if there is a notification confirming the hotkey execution. Would make sense to have:
1) turn on top-up
2) revert back to previous limit
3) discharge
--- bonus --
stop charging when sleeping
stop charging when app closes
the other one that's there.. 

...Than I can get rid of the icon completely (since it takes up a ton of space)

closed 2022-07-31 by abeldemoz