AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #462: Laptop not charging after downgrade

agneevX, 2021-11-17:
Hello,

I installed the latest version of Al Dente on Monterey. 

However I downgraded to Catalina and now my laptop won't charge over the set limit when I was on Monterey. 

How do I revert this?

MatthiasKerbl, 2021-11-17:
> Hello,
> 
> I installed the latest version of Al Dente on Monterey.
> 
> However I downgraded to Catalina and now my laptop won't charge over the set limit when I was on Monterey.
> 
> How do I revert this?

Hi @agneevX,

Which MacBook model do you have?

Cheers,
Matthias

agneevX, 2021-11-17:
This is a 2015 MBP.

MatthiasKerbl, 2021-11-17:
> This is a 2015 MBP.

I see, this should fix it:

1. Download AlDente Classic 2.2, this version should run on Catalina:https://github.com/davidwernhart/AlDente/releases/download/2.2/AlDente_2.2_Notarized.app.zip
2. Install AlDente Classic 2.2
3. Enable "Use Classic SMC Key(Intel)" and set the charge limit to 100%
4. Wait until your MacBook is fully charged
5. Uninstall AlDente again

Hope that helps!

Cheers,
Matthias

closed 2021-11-17 by MatthiasKerbl