AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #465: Battery Charge Limit suggestion

tonler1188, 2021-11-18:
Hi all,

 90% of the time my 15inch Macbook Pro 2018 is plug-in for my work. 
What battery charge limit would you suggest?  

50% (with biweekly battery calibration)
60% (with biweekly battery calibration)
80% (with biweekly battery calibration)

Any constructive suggestions are welcome! 
Thanks

MatthiasKerbl, 2021-11-18:
> Hi all,
> 
> 90% of the time my 15inch Macbook Pro 2018 is plug-in for my work. What battery charge limit would you suggest?
> 
> 50% (with biweekly battery calibration) 60% (with biweekly battery calibration) 80% (with biweekly battery calibration)
> 
> Any constructive suggestions are welcome! Thanks

Hi @tonler1188,

To be completely honest, I do not know either. The problem is that almost every MacBook model comes with a slightly different battery chemistry and even the same MacBook generation often has different battery suppliers which might result in a different charge limit being optimal for longevity. However, there is some data to work with. Apple limits charging to 80% which is a starting point but if you look into research papers that specifically look into LCO battery chemistries which is used for MacBook batteries, there is some data suggesting that even a lower percentage of 60 or 50% might be even better. Therefore, I would suggest with about 60% you should be close to the optimum. Just keep in mind that your battery calibration will get disturbed more quickly, the lower your charge limit is. Because of that, calibrating the battery at least biweekly is necessary.

Hope that helps!

Cheers,
Matthias

closed 2021-11-18 by MatthiasKerbl

tonler1188, 2021-11-18:
> Hi @tonler1188,
>
> To be completely honest, I do not know either. The problem is that almost every MacBook model comes with a slightly different battery chemistry and even the same MacBook generation [...]

Thanks for your quick reply and the info.  I will def. Try the 60% charge limit and be sure to run the batter calibration at least biweekly! 

Thank you again Matthias.

DamonHD, 2021-11-18:
> Because of that, calibrating the battery at least biweekly is necessary.

On that front, could you please enable some alternative for Intel Mode users such as me that automatically allows a full charge weekly, or whatever seems appropriate, if that will help (ie poor-man's calibration).

I can turn this into an issue if you wish.

Rgds

Damon

MatthiasKerbl, 2021-11-18:
> On that front, could you please enable some alternative for Intel Mode users such as me that automatically allows a full charge weekly, or whatever seems appropriate, if that will help (ie [...]

Hi @DamonHD,

Since most of our users have a MacBook from 2016 or newer we are currently not working on improving the functionality of AlDente for Intel Mode only users. However, I also do not want to exclude it from a potential update in the future.

Hope you understand that.

Cheers,
Matthias