AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #466: After update 1.01 issues to startup

Extrizyo, 2021-11-18:
Hello ,

I'm using AIDente PRO version for 1 month now without issues. But yesterday i did the update to 1.01 an now i'm having some issues.

Everytime i want to start the app i get the popup to disable optimized battery charging in macos. I checked this 10 times but it is off. I keep getting this popup. Any1 has a solution? 

Using this app on a new 14 inch MBP 

Greetings
Thomas

MatthiasKerbl, 2021-11-18:
Hi @Extrizyo,

I am sorry you are having issues with AlDente Pro.

It sounds like that the installation of AlDente Pro is somehow disturbed. I would recommend you to follow this uninstallation guide (https://apphousekitchen.com/how-to-uninstall-aldente-pro/) and after that try to do a fresh install of AlDente Pro again. You can download AlDente Pro 1.10 from here: https://github.com/davidwernhart/AlDente/releases/download/1.10/AlDente1.1.dmg

Hope that helps!

Cheers,
Matthias

Extrizyo, 2021-11-18:
Fixed. Thank you !

closed 2021-11-18 by MatthiasKerbl