AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #469: Prevent AlDente from showing settings when mac starts?

mjbaldwin, 2021-11-19:
Love AlDente!

But recently I've been noticing that every time I restart my Mac (M1 Air, Monterey), it shows its popup as if I'd clicked on it, although I hadn't:

[image]

(I hide the menubar icon using [Hidden Bar](https://apps.apple.com/us/app/hidden-bar/id1452453066?mt=12) which is why the popup doesn't appear linked to the icon.)

It's slightly annoying to have to click to dismiss it every time I start my Mac, and I don't remember this having always been the case. Is this behavior by design? And if so, is there a way to disable it? I know it's running, I don't need to be reminded every time I restart...

To be clear, I want AlDente to start when my Mac starts and have its icon in the menubar... I just don't want the additional popup to appear.

Thanks!

vincentpun, 2021-11-20:
It's a new feature that can't be turned off, it looks like... I'd appreciate it if there was a way to!

MatthiasKerbl, 2021-11-20:
Hi @mjbaldwin,

Due to some people having trouble finding the controls of AlDente we now show the Charge Limiter settings every time AlDente restarts. However, we understand that it is not necessary for everyone that’s why we will implement a setting to turn it off with the next update.

Cheers,
Matthias

mjbaldwin, 2021-11-20:
Thanks so much @MatthiasKerbl that's great to hear! And thanks again for the awesome app.

ammarfatani, 2021-11-22:
> Love AlDente!
>
> But recently I've been noticing that every time I restart my Mac (M1 Air, Monterey), it shows its popup as if I'd clicked on it, although I hadn't:
>
> [image]
>
> (I hide the menubar icon [...]

I agree. After a couple of times, it becomes annoying. Also, for some reason when it does show up it always be in the very far left of the screen underneath the Apple logo.

MatthiasKerbl, 2021-12-11:
Hi everyone,

We just released a new version of AlDente which allows you to disable the automatic popup at launch.

Cheers,
Matthias

closed 2021-12-11 by MatthiasKerbl