AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #472: After using the deactivate sleep mode function generally no longer sleep

burnout-syndrome, 2021-11-21:
I used the "Deactivate sleep" function for the first time today. After switching off, I noticed that the screen saver was no longer activated on my MacBook, the sleep item in the Apple menu is grayed out. A pmset-g in the terminal tells me "sleepdisabled 1" and displaysleep "display sleep prevented by com.apphousekitchen.aldente". How do I get back to my previous settings? Deactivating the slider in AlDente does nothing.

ShapeSh1ft3r, 2021-11-29:
@burnout-syndrome, Same thing happened to me. My MacBook discharged from 50 to zero with closed lid from Saturday to Monday (without charging).

My solution:
1. Turn off "Disable sleep until Charge Limit"
2. Uninstall AiDente Helper -> at this point you can put to sleep your machine again.
3. Reinstall AiDente (I don't turn on the Disable Sleep... setting again)

Hope this helps.

My Machine: MacBook Pro 14, 12.0.1, latest AiDente

MatthiasKerbl, 2022-02-02:
Hi everyone,

We have just released a new version of AlDente (1.14) which should fix the issues you experienced.
If you still encounter any of the issues after updating, please report back.

Download AlDente 1.14 here (https://github.com/davidwernhart/AlDente/releases) or update within the app.

Thank you for your patience.

Cheers,
Matthias

closed 2022-02-02 by MatthiasKerbl

lgberro, 2022-02-08:
`com.apphousekitchen.aldente-pro.helper` is preventing sleep when locked (the screen doesn't turn off). Using:
- AlDente 1.14
- Monterey 12.1 (21C52)
- MacBook Air (M1, 2020)

After killing the helper it sleeps normally.

[image]

MatthiasKerbl, 2022-02-11:
> `com.apphousekitchen.aldente-pro.helper` is preventing sleep when locked (the screen doesn't turn off). Using:
> 
> * AlDente 1.14
> * Monterey 12.1 (21C52)
> * MacBook Air (M1, 2020)
> 
> After killing the helper it sleeps normally.
> 
> [image]

Hi @lgberro,

It would be helpful if you could generate and share a debug file with us when it happens the next time. You can find a guide for the debug file here: https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/[](https://user-images.githubusercontent.com/33470003/153661181-e7187ca1-3d1e-4bce-93b8-999e961f399d.png)

Is the Sleep option greyed out when you encounter this problem?
grey sleep

Cheers,
Matthias