AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #475: AlDente keeps crashing

Chrizzzzle, 2021-11-24:
Hello there,

Today I tried to install AlDente 1.10 on my new 2021 Macbook Pro 14 (M1 Pro).
While the installation worked fine, the app crashes after clicking on the icon in the menu bar.
The loading icon appears when I move my mouse over the pop-up, and the app does not react to my clicks. I can't close the pop-up once it's opened.
I deactived "Optimized Battery Charging" in Settings and tried reinstalling the app.

Thank you for your help! :-)

Chrizzzzle, 2021-11-24:
PROBLEM SOLVED after updating to 12.0.1

nights62, 2021-11-29:
I am having the same problem on a Macbook Pro 13in M1 running 12.0.1 as soon as I move slider I get the spinning beach ball and I have to kill the app.

herwart, 2021-11-29:
I also have the same problem:  MacBook Pro (16", 2021), Apple M1 Max, macOS 12.0.1. But the window freezes after changing the percentage slider. The freeze of the AlDente window has side effects. For example, the screen sharing of MS Teams freezes too.

JorshyBitz, 2021-12-03:
I have the same issue as well. MBP 14' M1 Max I will do some troubleshooting and come back to the thread if I find a fix. If I adjust the slider at all it freezes and displays the spinning beach ball (the ball of death).

JorshyBitz, 2021-12-03:
This was shown in my console log along with helper SMC errors 

Prompting policy for hardened runtime; service: kTCCServiceAppleEvents requires entitlement com.apple.security.automation.apple-events but it is missing for accessing={<TCCDProcess: identifier=com.apphousekitchen.aldente-pro, pid=1084, auid=501, euid=501, binary_path=/Applications/AlDente.app/Contents/MacOS/AlDente>}, requesting={<TCCDProcess: identifier=com.apple.appleeventsd, pid=472, auid=55, euid=55, binary_path=/System/Library/CoreServices/appleeventsd>},

JorshyBitz, 2021-12-06:
figured out a way for it to work. if you dont use the slider it never crashes just manually type the number in until its fixed :)

Jason3022, 2021-12-11:
16 days have passed and still no official response?

JorshyBitz, 2021-12-11:
I know.... kind of weird. Might uninstall for now until this is fully sorted out seems a bit sus

MatthiasKerbl, 2021-12-11:
Hi everyone,

Thank you for reporting this issue.
Unfortunately, we are currently not able to reproduce it at all.
Does anyone use AlDente Pro or do you all have AlDente Free?

MatthiasKerbl, 2021-12-11:
> This was shown in my console log along with helper SMC errors
>
> Prompting policy for hardened runtime; service: kTCCServiceAppleEvents requires entitlement com.apple.security.automation.apple-events [...]

Hi @JorshyBitz,

Could you share your crash report file?

JorshyBitz, 2021-12-11:
> Hi @JorshyBitz,
>
>
>
> Could you share your crash report file?

I shared the error above. Im new to Mac coming from Windows how would I share the crash report file?

MatthiasKerbl, 2021-12-11:
> > > This was shown in my console log along with helper SMC errors
> > 
> > 
> > > 
> > 
> > 
> > > Prompting policy for hardened runtime; service: kTCCServiceAppleEvents requires entitlement com.apple.security.automation.apple-events but it is missing for accessing={<TCCDProcess: identifier=com.apphousekitchen.aldente-pro, pid=1084, auid=501, euid=501, binary_path=/Applications/AlDente.app/Contents/MacOS/AlDente>}, requesting={<TCCDProcess: identifier=com.apple.appleeventsd, pid=472, auid=55, euid=55, binary_path=/System/Library/CoreServices/appleeventsd>},
> > 
> > 
> > Hi @JorshyBitz,
> > Could you share your crash report file?
> 
> Im new to Mac coming from Windows how would I share the crash report file?

No worries, you can find the crash report file in Macintosh HD/Library/Application Support/Crash Reporter

cryptoAlgorithm, 2021-12-14:
Well, I can show a screen recording of the hang (I dont get a crash)

<video src="https://user-images.githubusercontent.com/64193267/146027068-51f1dcad-2253-4125-8924-4cf32093bbde.mov" controls></video>
For some reason the beach ball doesn't show up in the recording, but basically from 3s onwards it's hang.

cryptoAlgorithm, 2021-12-14:
It can only be resolved by force quitting from activity monitor or killing thru the terminal

MatthiasKerbl, 2021-12-14:
> Well, I can show a screen recording of the hang (I dont get a crash)
> 
>  Screen.Recording.2021-12-14.at.11.21.49.PM.mov 
> 
> For some reason the beach ball doesn't show up in the recording, but basically from 3s onwards it's hang.

Thanks for sharing, we are currently investigating this issue.

JorshyBitz, 2021-12-14:
> No worries, you can find the crash report file in Macintosh HD/Library/Application Support/Crash Reporter

I have nothing in crash reporter. keep in mind the application does work when I dont use the slider to adjust the batter percentage. if I type it in it works just fine.

JorshyBitz, 2021-12-14:
> Hi everyone,
> 
> Thank you for reporting this issue. Unfortunately, we are currently not able to reproduce it at all. Does anyone use AlDente Pro or do you all have AlDente Free?
> 
> It would be really helpful if someone could share a crash report file with us. You can find the crash reports in Macintosh HD/Library/Application Support/Crash Reporter.

What apple device are you using this on?

herwart, 2021-12-14:
> > Hi everyone,
> > Thank you for reporting this issue. Unfortunately, we are currently not able to reproduce it at all. Does anyone use AlDente Pro or do you all have AlDente Free?
> > It would be really helpful if someone could share a crash report file with us. You can find the crash reports in Macintosh HD/Library/Application Support/Crash Reporter.
> 
> What apple device are you using this on? I am using the free version btw

AlDente Free, MacBook Pro 16", Apple M1 Max
Last crash is about 10 day ago. Feels more stable now. I have installed the latest update.
Always when it crashed, the MacBook was connected to power.
There is no specific AlDente crash report, only the common "intervals…plist". Do you need this? Please tell me how to share this file securely as it contains a lot of information about my Mac.

MatthiasKerbl, 2021-12-15:
> AlDente Free, MacBook Pro 16", Apple M1 Max Last crash is about 10 day ago. Feels more stable now. I have installed the latest update. Always when it crashed, the MacBook was connected to power. [...]

Hi @herwart,

Thank you for the information.
We could find the bug and will fix it with the next update.
No need for any crash reports.

Cheers,
Matthias

JorshyBitz, 2021-12-15:
Great!

rippeddisc, 2021-12-18:
Same issue.  Beach-ball when using slider.  Updated app and os 12.1

closed 2021-12-20 by MatthiasKerbl

thirteenkai, 2024-11-12:
AlDente crashes on MacBook M4