AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #476: Discharge Mode cycles between charging, not charging and discharging in macOS Monterey on M1 Air

Havhingstor, 2021-11-25:
I have a MacBook Air with the M1-chip, which runs on macOS Monterey 12.0.1.
I use AlDente Free and my Limit is set to 70%. I am connected to a docking station, which charges my MacBook. My Battery is at 78% and I want to discharge the Macbook using discharge mode. If I activate discharge mode, it will discharge for a few seconds, then will say "Connected, but not charging" for a few seconds and then will charge for around one second. Then the cycle will repeat. I use AlDente 1.10

The problem isn't very urgent, but I wanted to report it since i haven't seen anyone talk about this in Monterey.

Here is a short video of my problem:

https://user-images.githubusercontent.com/59767224/143486069-e83ec92c-f38a-4d70-b1b9-395fb4c9fcc5.mp4

MatthiasKerbl, 2021-11-25:
Hi @Havhingstor,

Do you have "Optimized battery charging" disabled in System Preferences?

Cheers,
Matthias

Havhingstor, 2021-11-25:
I've reinstalled AlDente and currently it's working

closed 2021-11-25 by MatthiasKerbl