AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #477: -20% Black Friday Sale on AlDente Pro!

MatthiasKerbl, 2021-11-26:
Hi everyone,

If you are interested in AlDente Pro, you can get it now 20% cheaper for a limited time only.

What’s AlDente Pro?
AlDente Pro is our paid version of AlDente. It has many more features such as Heat Protection, Sailing Mode, Top Up, Calibration Mode,... It offers a better design and has live status icons. If you are interested in getting the most out of your battery, check out our website: https://apphousekitchen.com/

Cheers,

David and Matthias

sqleonhart, 2021-11-27:
@MatthiasKerbl  Hi Matthias, I just made a purchase but did not get any email (apart from Paypal payment confirmation). Not sure what to do. Could you please help?

MatthiasKerbl, 2021-11-27:
> @MatthiasKerbl Hi Matthias, I made a purchase a couple hours ago but did not get any email (apart from Paypal payment confirmation). Not sure what to do. Could you please help?

Hi @sqleonhart,

I am sorry for the inconvenience. Which email address did you use?

Cheers,
Matthias

sqleonhart, 2021-11-27:
Hi @MatthiasKerbl

Just made the email I used visible on my @sqleonhart profile. Please let me know if you could see it.

Regards
Andrew

MatthiasKerbl, 2021-11-27:
> Hi @MatthiasKerbl
> 
> Just made the email I used visible on my @sqleonhart profile. Please let me know if you could see it.
> 
> Regards Andrew

Thanks, I just checked your purchase at our payment provider and everything seems normal. However, I sent the purchasing emails again and if there is a problem again, I will also send you an email with your license key from our support email address.

Cheers,
Matthias

sqleonhart, 2021-11-27:
@MatthiasKerbl I got the email from your support address and activated the app, many thanks. But the purchasing emails didn't seem to reach me again (I've checked my junk mail folder too). Not sure if it's just my email address... but all good for me now. Thanks again.

jellyworks, 2021-11-29:
I bought aldente a few days ago using my credit card. But I haven't received any emails. The email used is yeong.rok.lee@gmail.com.

MatthiasKerbl, 2021-11-29:
> 

Hi @jellyworks,

You had a typo in your email address, I have sent you the license key to the correct email address now.

Cheers,
Matthias

closed 2021-11-30 by MatthiasKerbl

dactylo, 2021-12-06:
Is it too late for the BF/Cyber Monday sale?

Also could someone clarify whether the menubar icon of the Pro version looks like a plate of spaghetti? I like to keep food away from my computer and I'd prefer something more abstract. Thanks.

MatthiasKerbl, 2021-12-06:
> Is it too late for the BF/Cyber Monday sale?
> 
> Also could someone clarify whether the menubar icon of the Pro version looks like a plate of spaghetti? I like to keep food away from my computer and I'd prefer something more abstract. Thanks.

Hi @dactylo,

Unfortunately, the Black Friday Sale is already over.
The menubar icon in the AlDente Pro is customizable. You have the option of live status icons as described on our website(https://apphousekitchen.com/) but you have also the option of just using the battery percentage. With the next update, you will have the option to completely deactivate the menubar icon as well.

Cheers,
Matthias

samwzlim, 2021-12-10:
@MatthiasKerbl

Other than the Black Friday Sale, do you offer any other sales for AlDente? I wanted to purchase AlDente Pro [9 days ago](https://github.com/davidwernhart/AlDente/issues/479#issuecomment-983403796), but unfortunately, I was a few days late to the party. I am using AlDente free all the time, and would like to get AlDente Pro for the additional features, but wish I can get it at a lower price.

MatthiasKerbl, 2021-12-11:
> @MatthiasKerbl
>
> Other than the Black Friday Sale, do you offer any other sales for AlDente? I wanted to purchase AlDente Pro [9 days [...]

Hi @samwzlim,

Thank you for your interest in AlDente Pro and I am glad you like AlDente free!
Unfortunately, the Black Friday Sale is over and we do not offer any other kind of discount currently.

Cheers,
Matthias

CollinGrol, 2023-08-24:
Hi @MatthiasKerbl 

I have your app now for some time and have also thought about upgrading to the Pro version. Then I found this GitHub chat and saw that you have made almost two years ago a -20% Black Friday Sale. Therefore I wanted to ask if there is plane  a discount in the nearer future?

Regards 
Collin

MatthiasKerbl, 2023-08-25:
Hi @CollinGrol,

Unfortunately, there are no plans for a discount in the near future. The only discount that's available right now is the student discount at [StudentAppCentre](https://www.studentappcentre.com/App/AlDente).

Best,
Matthias

VestigiaLunae, 2024-11-28:
Will there be a discount tomorrow?