AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #479: Is there a student discount?

jesusdeveloper, 2021-11-27:
Hello!

I know there is a 20% discount currently for Black Friday but I was wondering if there is any greater discount if you are a college student.

Thank you.

MatthiasKerbl, 2021-11-27:
> Hello!
> 
> I know there is a 20% discount currently for Black Friday but I was wondering if there is any greater discount if you are a college student.
> 
> Thank you.

Hi @jesusdeveloper,

We do not offer a student discount. The 20% Black Friday discount is the only one we offer.

Cheers,
Matthias

closed 2021-11-27 by MatthiasKerbl

jesusdeveloper, 2021-11-27:
> > Hello!
> > I know there is a 20% discount currently for Black Friday but I was wondering if there is any greater discount if you are a college student.
> > Thank you.
> 
> Hi @jesusdeveloper,
> 
> We do not offer a student discount. The 20% Black Friday discount is the only one we offer.
> 
> Cheers, Matthias

ok! Thank you Matthias!

samwzlim, 2021-12-01:
Hi @MatthiasKerbl, I am a student myself and am interested in purchasing Aldente Pro in order to preserve the battery health of my MacBook. Unfortunately, I missed the Black Friday sale as I just received my MacBook today. Any chance I can still benefit from the discount, since it was just live a few days ago?