AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #482: "Disable Sleep until Charge Limit" keeps disabling itself

dolinod, 2021-12-01:
I've noticed that sometimes when I check AlDente settings that "Disable Sleep until Charge Limit" has become disabled even though I've previously enabled it. I haven't explored enough to consistently reproduce this on demand; however, it has reproduced itself multiple times.

koolman2, 2021-12-05:
I'm having the opposite issue. Mine keeps enabling itself. I plug in my computer, lock the screen, and wait for the screen to shut off. Every time I unplug, I have to go back in and toggle the setting on and back off for it to allow the screen to shut off. To be clear, the computer still charges if the screen turns off while locked.

MatthiasKerbl, 2022-01-07:
Hi @dolinod and @koolman2,

Thank you for reporting this issue and sorry for the inconvenience.
We think that we have found the cause of this issue and it will be fixed with the next update.

Cheers,
Matthias

MatthiasKerbl, 2022-01-16:
Hi there,

We have just released a new version of AlDente with a fix for this issue.
You can download the newest version [here](https://github.com/davidwernhart/AlDente/releases) or just do the update within the app.

Cheers,
Matthias

closed 2022-01-16 by MatthiasKerbl