AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #483: Doesn't Charge Past 1%

abeldemoz, 2021-12-02:
AlDente-Pro user here. I used my MacBook Pro away from the charger until it ran out of battery. When I pugged it in, the battery was stuck at 1% for over an hour. As soon as I quit AlDente-Pro, it started charging properly. It was a miracle that I got the Mac to turn on at 1% battery, it was showing me a black screen with the red battery.

[image]

abeldemoz, 2021-12-20:
Can somebody respond to this please?!?!?!

MatthiasKerbl, 2021-12-21:
Hi @abeldemoz,

We have never seen an issue like this before. I have a couple of ideas of what could be the issue but I am not sure if this is related to AlDente. Furthermore, when google-searched for this bug, a lot of posts about software and hardware defects appear and most of them are older than AlDente.

To start with, please give us more information about your setup:
1. Which MacBook model do you have?
2. Which MacOS version do you use?
3. Which version of AlDente Pro do you have installed?
4. How old is your MacBooks battery?
5. How many charge cycles does your battery have?
6. How much capacity does your battery still have when you fully charge it?
7. What does the macOS battery status say?
8. When was the last time you completely drained your battery to 0%?
9. When was the last time you used the Calibration Mode of AlDente or did a full battery cycle of 100%-0%-100%?

Cheers,
Matthias

abeldemoz, 2021-12-24:
I think the issue might be something to do with the Mac going into sleep mode or hibernation mode. A few nights ago, I noticed a similar issue. I closed my MacBook Pro and left it on my bedside table unplugged overnight. In the morning, I connected the MacBook Pro to my monitor via USB-C without opening the lid and proceeded to use my Mac in clamshell mode. It was on 24% battery when I plugged it in to the monitor, and after an hour it stayed at 24% and wouldn't charge despite the fact that I set the AlDenete-Pro charging limit to 65%. Once I quit AlDente-Pro and reopened it, the Mac started charging as normal and charged past 24%.

1. 2020 4-port Intel 13-inch MacBook Pro 2.0Ghz i5 16GB RAM 512GB storage
2. macOS 12.1
3. AlDente Pro 1.12
4. battery manufactured 10 June 2020, unboxed 31 July 2020
5. 103 charge cycles
6. 88% battery health,  4,480 mAh remaining capacity
7. battery status/condition is normal
8. I drain the battery to 0% and charge up back up to 100% on the first day of every month to recalibrate my battery
9. I drain the battery to 0% and charge up back up to 100% on the first day of every month to recalibrate my battery

MatthiasKerbl, 2022-01-05:
Hi @abeldemoz,

Thank you for providing all the information.
Unfortunately, I do not really know what exactly the issue is or how to solve it. One theory is that your battery is heavily miscalibrated but since you are doing at least one full cycle every month that should not be the reason. Furthermore, degradation should not be an issue either since your battery is still relatively new and healthy. Other reasons could be battery or MacBook hardware defects or macOS or AlDente software defects. Another theory is that it got discharged too deeply and needed time to get to a SOC where it could restart. How long did you leave it unplugged from running out of battery until you plugged it in again?

Do you still experience this issue when you discharge your MacBook completely?

closed 2022-02-25 by MatthiasKerbl