AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #489: [Feature Request] Show temperature in menubar side by side with battery percent.

Eric-Guo, 2021-12-04:
Would be great if can display the temperature at the same time of battery percent, like up/down style.

Before (network speed is another app, not relative with AIDente): 

[image]

After
[image]
:

Stickyhavr, 2022-03-20:
Yes. I second the request. I came here to start a new feature request but saw that my requests already exist in this post (#489) and also #620. :-)

basmets6, 2022-04-17:
+1 for me as well. Would be super useful if you actively use the "Heat Protection" feature.

I purchased the full version, and a couple of times my MBP just wasn't charging up to the limit.
A bit stumped, I figured out that it was due to the Heat Protection.. Then I opened the settings menu and waited, waited, waited until temp dropped.

Would therefore be super useful to have this in the menu bar, so you don't have to keep that window open the entire time.