AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #491: Al Dente broke my Magsafe Charger (Blinking Orange) 2021 M1 Macbook 14"

monoglotter, 2021-12-05:
**Describe the bug**
Magsafe charger worked beautifully every single time. Plugged in, and started charging. After installing Al Dente, it started blinking sporadically and didn't charge even when the limit was set at 80%. It just kept going down and down and the charger didn't charge. Kept blinking orange.

I tried uninstalling Al Dente, shutting down then turning computer back on, restarting, all of this did not work.

I followed uninstall instructions to set charge limit to 100%. Also, charges fine on USB C. Magsafe still blinks orange when plugged in. 

I believe there's no way to reset SMC on M1 Macbooks, so not sure what to do now.

**Desktop (please complete the following information):**
 - MacBook model (https://support.apple.com/en-us/HT201608)
 M1 Macbook Pro 14" 2021
 - macOS version (https://support.apple.com/en-us/HT201260)
 OS X Monterey 12.0.1
 - AlDente Pro or AlDente Free?
 Free
 - AlDente version
 Latest
 - List of all enabled/disabled settings in AlDente
N/A

ovq, 2021-12-09:
@monoglotter 
Problems with MagSafe charging seems to be a known issue on 12.0.1 and should be fixed in the upcoming 12.1. 
https://www.macworld.com/article/556564/apple-macbook-pro-magsafe-charging-closed.html

monoglotter, 2021-12-09:
> @monoglotter Problems with MagSafe charging seems to be a known issue on 12.0.1 and should be fixed in the upcoming 12.1. https://www.macworld.com/article/556564/apple-macbook-pro-magsafe-charging-closed.html

Yes, I saw that too after researching. I'm thinking its an Al Dente problem though, since it only started happening once I installed Al Dente and started using it. (Luckily, USB C charging works fine with Al Dente.)

ovq, 2021-12-09:
> Yes, I saw that too after researching. I'm thinking its an Al Dente problem though, since it only started happening once I installed Al Dente and started using it. (Luckily, USB C charging works [...]

Noted. 
12.1 is in RC state, you should be able to check soon, if this fixes the problem.

monoglotter, 2021-12-09:
> 

Would you happen to know when 12.1 is coming out? I'd be willing to try it to see if it fixes the problem.

monoglotter, 2021-12-09:
> Noted. 12.1 is in RC state, you should be able to check soon, if this fixes the problem.

Would you happen to know when 12.1 is coming out? I'd be willing to try it to see if it fixes the problem.

ovq, 2021-12-09:
> Would you happen to know when 12.1 is coming out? I'd be willing to try it to see if it fixes the problem.

I don't. However, it doesn't take long after RC state for Apple to release it to the public - unless they find a major showstopper in the code... :-) 

I expect it to drop end of this / early next week.

amplicity, 2021-12-19:
@monoglotter just curious if you ever got this resolved?

monoglotter, 2021-12-19:
I did not get it resolved yet.

jotka, 2021-12-29:
@monoglotter has this been resolved in 12.1?

monoglotter, 2021-12-29:
I was actually about to update this post.

So the first MacBook that I made this post with works with the MagSafe
charger now after updating to 12.1. I installed AlDente when it was 12.0.1.

However, I just got a new (2nd) MacBook and upgraded to 12.1. Then, I
installed aldente. On this MacBook, the charger doesn’t work. Not sure what
is going on.

KexianShen, 2022-01-01:
> I was actually about to update this post. So the first MacBook that I made this post with works with the MagSafe charger now after updating to 12.1. I installed AlDente when it was 12.0.1. However, [...]

thanks for your report.

MatthiasKerbl, 2022-01-15:
> I was actually about to update this post. So the first MacBook that I made this post with works with the MagSafe charger now after updating to 12.1. I installed AlDente when it was 12.0.1. However, [...]

Hi @monoglotter,

Could you describe in more detail what exactly the problem is?
Did the Magsafe charger stop working completely after installing AlDente?

closed 2022-02-25 by MatthiasKerbl