AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #492: AlDente Free : Discharge button missing

rramchar, 2021-12-06:
At some point the Discharge button in the UI disappeared.
After that my 2018 MBP doesn't stay at 80% and charges to 100%

MatthiasKerbl, 2021-12-06:
Hi @rramchar,

Do you have the Intel Mode activated?

Cheers,
Matthias

rramchar, 2021-12-06:
Yes I do.

rramchar, 2021-12-06:
Attached pictures
[image]
[image]

simulatxd, 2021-12-08:
hi, btw iI have same issue, I should turn off intel mode to make appear discharge button.

[image]

rramchar, 2021-12-08:
@simulatxd does the Discharge button immediately show up when you disable "Intel mode"?

rramchar, 2021-12-08:
Had someone point out to me this FAQ entry

https://apphousekitchen.com/feature-explanation-intel-mode/

Intel mode doesn't mean "for Intel MacBooks" as I incorrectly assumed.  Disabling "Intel Mode" makes the discharge button visible and the MacBook is currently discharging. 

Closing this issue.

closed 2021-12-08 by rramchar