AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #493: Sailing mode not honored after restart

ovq, 2021-12-06:
**Describe the bug**
Charge Limit:  80%
Sailing mode: enabled 10% 
current battery level  at 77%

After a restart, the Mac will charge to 80%, Al Dente will not block charging until 70% is reached.

**To Reproduce**
-- 

**Expected behavior**
After a restart, Al Dente should respect sailing settings and let the battery drain until threshold reached. 

**Screenshots**
-- 

**MacBook (please complete the following information):**
 - MacBook Pro 14" 
 - macOS version 12.1 beta 4 
 - AlDente Pro or AlDente Free?
 - AlDente Pro 1.10
 - List of all enabled/disabled settings in AlDente:
- 
**Additional context**

MatthiasKerbl, 2021-12-11:
Hi @ovq,

Thanks for pointing this out, we might change that in one of the next updates.

Cheers,
Matthias

MatthiasKerbl, 2022-01-16:
Hi there,

We have just released a new version of AlDente with a fix for this issue.
You can download the newest version [here](https://github.com/davidwernhart/AlDente/releases) or just do the update within the app.

Cheers,
Matthias

closed 2022-01-16 by MatthiasKerbl