AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #497: AlDente Discharge mode toggles between charging and discharging

cryptoAlgorithm, 2021-12-13:
**Describe the bug**
When discharge mode is activated, macOS indicates charger disconnection after a few seconds (expected), but after some time, the battery changes back to charging and the display brightness increases, as though discharge mode is deactivated. However, the discharge button is activated throughout this time. After a few seconds, discharge mode suddenly works as expected again. This cycle repeats for as long as I have tested the discharge feature (prob 5 mins).

**To Reproduce**
Steps to reproduce the behavior:
1. Open the AlDente menu bar item
2. Click on the Discharge button
3. Wait

**Expected behavior**
Discharge mode continuously discharges

**Screenshots**
[image]

**MacBook (please complete the following information):**
 - MacBook model: MacBook Pro (14-inch, 2021)
 - macOS version: 12.0.1
 - AlDente Pro or AlDente Free: AlDente Free
 - AlDente Version: 1.11, latest as of now
 - List of all enabled/disabled settings in AlDente: Freshly reinstalled, all default settings except for:
 - * Discharge mode active
 - * Charge limit set to 50%

**Additional context**
Add any other context about the problem here.
My battery is fully charged, and this bug doesn't happen when the AlDente charge limit is set to 100% (happens at any other percentage)

cryptoAlgorithm, 2021-12-13:
For clarification: the MagSafe charging cable wasn't touched throughout the video of the bug occurring

MatthiasKerbl, 2021-12-13:
Hi @cryptoAlgorithm,

Thank you for reporting this bug.
Do you have Apple's "Optimized battery charging" disabled in settings?

Cheers,
Matthias

ericloong90, 2021-12-17:
Can confirm on my 16" M1 Max MBP as well.

MacBook (please complete the following information):

MacBook model: M1 Max 32GB MacBook Pro (16-inch, 2021)
macOS version: 12.0.1
AlDente Pro or AlDente Free: AlDente Pro
AlDente Version: 1.11, latest as of now
List of all enabled/disabled settings in AlDente: Freshly reinstalled, all default settings except for:
Discharge mode active
Charge limit set to 80%

THEWHITEBOY503, 2021-12-21:
Same issue here. 2020 MBP M1 13". Charge limit also does not LIMIT charging, but rather slow it down once it reaches X percent. IE: it will stop charging, then start again, then stop, start, etc. Same deal with discharge mode, but it goes between discharging and connected but not charging. 

MacBook model: M1 16GB MacBook Pro (13-inch, 2020)
macOS version: 12.0.1
AlDente Pro or AlDente Free: AlDente Free
AlDente Version: 1.12, latest as of now
List of all enabled/disabled settings in AlDente: all default settings except for:
Discharge mode active
Charge limit set to 50%

fuzzy-logic, 2022-01-17:
Can confirm also getting same behaviour on discharge, with the addition of screen constantly dimming / un-dimming at a rate faster (every ~3-4s) then the charge/discharge (every ~8-10s) rate. This only started happening after upgrade to Al Dente 1.13. Happens on Magsafe or USB-C power

MacBook model: M1 32GB MacBook Pro (13-inch, 2020)
macOS version: 12.1
AlDente Pro or AlDente Free: AlDente Pro
AlDente Version: 1.13

MatthiasKerbl, 2022-03-02:
Hi @cryptoAlgorithm,

Do you still experience this issue with our latest version AlDente 1.14?

Cheers,
Matthias

cryptoAlgorithm, 2022-05-08:
No, it appears to be resolved. Thanks!