AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #500: verification of Pro subscription fails, error message states there is no internet connection

PCDube, 2021-12-13:
[image]
[](url)
[image]

Hello.  I'm having an issue with Aldente not being able to verify my Pro license.  I get this message, even when I am connected to the internet.  This just started today.  I have restarted several times.  I have quit and restarted Aldente several times also.

Peter

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.

**MacBook (please complete the following information):**
 - MacBook model (https://support.apple.com/en-us/HT201608)
 - macOS version (https://support.apple.com/en-us/HT201260)
 - AlDente Pro or AlDente Free?
 - AlDente Version?
 - List of all enabled/disabled settings in AlDente:

**Additional context**
Add any other context about the problem here.

PCDube, 2021-12-13:
Forgot to add details!
[image]

[image]

MatthiasKerbl, 2021-12-13:
Hi @PCDube,

Thank you for reporting this issue we are currently investigating it.

A quick workaround should be as follows:

1. Go into Settings/General Settings
2. Click on "Manage License"
3. A popup window should open, click on "Deactivate License"
4. Click on "Activate License" and enter your email address and license code again

Everything should work fine again.

Cheers,
Matthias

PCDube, 2021-12-13:
Matthias,
the work around worked great!   Thanks!!!

MatthiasKerbl, 2022-03-03:
Hi @PCDube ,

We have just released AlDente 1.15 which fixed the issue you mentioned.
You can just do the update in AlDente or download the newest version here: https://github.com/davidwernhart/AlDente/releases

Cheers,
Matthias

closed 2022-03-03 by MatthiasKerbl

PCDube, 2022-03-03:
Thanks!  Downloaded, Installed and ReLaunched!