AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #501: No more option to remove icon from dock?

brypwn, 2021-12-14:
**Describe the bug**
Al Dente free no longer has the ability to remove the icon from the dock

**To Reproduce**
Steps to reproduce the behavior:
This behaviour can be observed on Al Dente 1.11

**Expected behavior**
Should be able to remove icon from dock?

**MacBook (please complete the following information):**
- M1 Macbook Air 2020, macOS Monterey 
- AlDente Free

closed 2021-12-14 by brypwn

markouustalu, 2021-12-15:
Aldente used to be only up there beside the clock, now it is also in dock. There is no option to remove it. Closing from dock closes it down completely.
[image]

MatthiasKerbl, 2021-12-15:
> Aldente used to be only up there beside the clock, now it is also in dock. There is no option to remove it. Closing from dock closes it down completely. [image]

Hi @markouustalu,

You can disable the dock icon in Settings/Appearance/Other "Show Dock Icon" disabled.

Cheers,
Matthias

markouustalu, 2021-12-15:
yes, indeed you can, and it works as needed. i guess the seven options on that page were too much for me to read and i didn't make it to the last one. Thank you again.

simon-forman, 2024-01-21:
It's grayed out and non interactable

MatthiasKerbl, 2024-01-23:
Hi @simon-forman,

the setting to deactivate the dock icon is also available in the free version here:
[image]

Best,
Matthias

Mshkss, 2025-07-21:
didnt worj