AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #503: Incorrect RU translation in the Sailing Mode section

dimasch, 2021-12-20:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Settings'
2. Click on tab 'Наcтройки заряда'
3. Scroll down to 'Парусный режим'
4. See incorrect translation `Пывёт до`

**Expected behavior**
`Плывёт до`

**Screenshots**
[image]

**MacBook (please complete the following information):**
 - MacBook Pro 14" M1 Pro (https://support.apple.com/kb/SP854?locale=en_US)
 - macOS Monterey 12.0.1
 - AlDente Pro
 - AlDente Version 1.11

**Additional context**
Used the RU locale

MatthiasKerbl, 2021-12-20:
Hi @dimasch,

Thank you for reporting this mistake. We will fix it with the next update.

Cheers,
Matthias

MatthiasKerbl, 2021-12-20:
Hi @dimasch,

we just released AlDente 1.12 which has a fix for the mistake you mentioned.
You can download the newest version here (https://github.com/davidwernhart/AlDente/releases) or just update your current version.

Cheers,
Matthias

closed 2021-12-20 by MatthiasKerbl

dimasch, 2021-12-20:
@MatthiasKerbl  I really appreciate it.

dimasch, 2021-12-20:
@MatthiasKerbl Missing space after words:
[image]

MatthiasKerbl, 2021-12-20:
> @MatthiasKerbl Missing space after words: [image]

Oh no haha
Will be fixed with the next release.